chipsalliance/firrtl #62: Lastet version of firrtl causes error with test TblSpec.scala

chick, 2016-02-11:
Using the latest versions of Chisel3 master: c85f9ff1c9ca9286d44a94f4a00aed67f4d916d3
and latest version of firrtl: 55ab528a3e8efaecf62163828f30cbd249623028
In chisel3 directory
```
sbt
> test-only chiselTests.TblSpec
```
In either master or from the merge associated with this pull request
I see errors like:
```
Starting High Form Check
Fatal Uncaught Exception.
NoFileInfo: [module Tbl]  Width cannot be negative or zero.
NoFileInfo: [module Tbl]  Width cannot be negative or zero.
TblTester5247337683591930572.fir@8.4: [module Tbl]  Width cannot be negative or zero.
   at /Users/chick/Aspire/IdeaProjects/firrtl/src/lib/stanza/compiler/lscore.lstanza@396.3
   at /Users/chick/Aspire/IdeaProjects/firrtl/src/lib/stanza/core/core.stanza@64.4
```

ucbjrl, 2016-02-11:
The firrtl error is:

    Reading TblTester3646275766733919895.fir
    Compiling!
    Starting Working IR
    Finished Working IR
    Milliseconds since start: 0
    Milliseconds for this pass: 0

    Starting High Form Check
    Finished High Form Check
    Milliseconds since start: 1
    Milliseconds for this pass: 1

    Starting Resolve Kinds
    No entry for key T_12 in hashtable
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/compiler/lscore.lstanza@396.3
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/core.stanza@64.4
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/verse.stanza@421.6
       at passes.stanza@662.44
       at ir-utils.stanza@589.47
       at passes.stanza@657.9
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/core.stanza@1351.20
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/core.stanza@1351.33
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/core.stanza@1351.33
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/core.stanza@1351.33
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/core.stanza@1352.3
       at ir-utils.stanza@600.24
       at passes.stanza@658.9
       at passes.stanza@665.6
       at passes.stanza@690.24
       at passes.stanza@697.12
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/core.stanza@1351.20
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/core.stanza@1352.3
       at passes.stanza@696.29
       at compilers.stanza@218.11
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/core.stanza@1339.12
       at /Volumes/vpbRoad/noArc/clients/ucb/git/ucb-bar/firrtl/src/lib/stanza/core/core.stanza@1342.3
       at compilers.stanza@215.15
       at compilers.stanza@205.3
       at firrtl-test-main.stanza@170.21
       at firrtl-test-main.stanza@177.0
       at NoFileInfo
    res = -1

Running make regress in the firrtl source directory seems to produce the same error.

ucbjrl, 2016-02-11:
It looks like commit bae6ba5d2405ba1caf7400f15d8875d4a79b8bae (More bug fixes) is the culprit - specifically removing some of the stanza passes: src/main/stanza/compilers.stanza

    @@ -50,9 +50,9 @@ public defmethod passes (c:StandardVerilog) -> List<Pass> :
       ;RemoveSpecialChars()     
       ;TempElimination()                 ; Needs to check number of uses
       ;===============
    -      CInferTypes()
    -      CInferMDir()
    -      RemoveCHIRRTL()            
    +      ;CInferTypes()
    +      ;CInferMDir()
    +      ;RemoveCHIRRTL()            
       ;===============
       ToWorkingIR()            

and it looks like Palmer has fixed this with 5902fe349fbadbe66b4553b286fbf2db74af96c2.

       ;===============

closed 2016-02-11 by ucbjrl