chipsalliance/firrtl #64: Difference in printf between scala and stanza firttl

chick, 2016-02-23:
There appears to be some difference between scala and stanza in the formatting of some
printf statements.  Here is a characteristic diff output, note the quotes around printf args
```
diff scala-firrtl.v stanza-firrtl.v
402c402
<             $fwrite(32'h80000002,"setting tbl(%h) to %h\n",io_load_routing_table_request_bits_addr,io_load_routing_table_request_bits_data);
---
>             $fwrite(32'h80000002,"setting tbl(%h) to %h\n","io_load_routing_table_request_bits_addr","io_load_routing_table_request_bits_data");
```

[files.zip](https://github.com/ucb-bar/firrtl/files/141965/files.zip)

chick, 2016-02-24:
I think this problem is the one currently breaking chisel3 chiselTests.Printf.scala

kammoh, 2016-02-24:
I wasn't able to observe the double-quoted node names you had in the printfs (using master). I have a dirty fix for the printf in my fork ( PR #70 ) which passes Chisel3 tests as well as handling printfs I had in my own design. Please try and see if it fixes this problem as well.

jackkoenig, 2016-02-25:
@kammoh's PR fixed the chiselTests.PrintfSpec problem, but this problem remains. It actually lead me to finding a new printf bug that I'll file in a minute. Anyway, this is a stanza bug, here's a small example:
```
circuit Bug :
  module Bug :
    input clk : Clock
    input a : UInt<1>
    input b : UInt<1>

    printf(clk, UInt<1>(1), "%d\n", a)
    printf(clk, UInt<1>(1), "%d\n", b)
    node c = and(a, b)
    printf(clk, UInt<1>(1), "%d\n", c)
```
Stanza FIRRTL incorrect output:
```Verilog
module Bug(
   input   clk,
   input   a,
   input   b
);
   wire  c;
   assign c = a & b;
   always @(posedge clk) begin
      `ifndef SYNTHESIS
         if(1'h1) begin
            $fwrite(32'h80000002,"%d\n","a");
         end
      `endif
      `ifndef SYNTHESIS
         if(1'h1) begin
            $fwrite(32'h80000002,"%d\n","b");
         end
      `endif
      `ifndef SYNTHESIS
         if(1'h1) begin
            $fwrite(32'h80000002,"%d\n","c");
         end
      `endif
   end
endmodule
```
Scala FIRRTL correct output
```Verilog
module Bug(
   input   clk,
   input   a,
   input   b
);
   wire  c;
   assign c = a & b;
   always @(posedge clk) begin
      `ifndef SYNTHESIS
         if(1'h1) begin
            $fwrite(32'h80000002,"%d\n",a);
         end
      `endif
      `ifndef SYNTHESIS
         if(1'h1) begin
            $fwrite(32'h80000002,"%d\n",b);
         end
      `endif
      `ifndef SYNTHESIS
         if(1'h1) begin
            $fwrite(32'h80000002,"%d\n",c);
         end
      `endif
   end
endmodule
```
As in @chick's original comment, Stanza FIRRTL is surrounding the $fwrite arguments in quotes.

azidar, 2016-04-09:
Has this been fixed? Can we close it?

jackkoenig, 2016-04-09:
Well there is still a difference between Scala and Stanza, but String Literals are handled properly in Scala Firrtl, and there's a PrintfSpec to test it.

closed 2016-05-02 by jackkoenig