chipsalliance/firrtl #65: No Support for Zero-width wires

jackkoenig, 2016-02-23:
See bug fix in b487ab24c0e298e7781f74811b8bfcd24384d1b5
This was due to rocket regression having memory of depth = 1, thus address width = 0. This makes sense because the address port is unnecessary in this case.

If I recall correctly, support for zero-width wires was mentioned as future work? But if they are supported in Chisel, I think it should be now work.

colinschmidt, 2016-02-23:
I would also like to see this feature implemented sooner rather than later!

azidar, 2016-02-23:
Yeah, the tricky thing is handling every corner case correctly. Off the top of my head - 

    wire w0 : UInt<0>
    wire w1 : UInt<1>
    wire w2 : UInt<1>

    add(w1,w0) // is the result width=1 or width=2?
    mul(w1,w0) // is the result width=1 or width=0?
    mux(w0,w1,w2) // => w2? or is it w1?
    mux(w1,w0,w2) // no clue

I think in general, a good rule of thumb is for arithmetic operations, its value should be treated as zero. I think the trickier part is determining the width inference rules.

ccelio, 2016-02-23:
Didn't Jim already handle zero-width wires in Chisel2? I don't think a complete reinvent of the wheel is needed on how different operations are handled.

A zero-width wire basically doesn't exist. So most things should just be an identity property, I would imagine. It's also fair to throw errors on some operations and allow others.

aswaterman, 2016-02-23:
The proposal isn't to change the semantics, but to get it working in
Chisel3, where the feature is currently missing.  This should require
nothing of Chisel3, but needs support in FIRRTL.

ucbjrl, 2016-02-23:
On Feb 23, 2016, at 11:39 AM, Adam Izraelevitz <notifications@github.com> wrote:

> Yeah, the tricky thing is handling every corner case correctly. Off the top of my head -
>
> wire w0 : UInt<0>
> wire w1 : UInt<1>
> wire w2 : UInt<1>
>
> add(w1,w0) // is the result width=1 or width=2? [...]

The approach I took in Chisel2 was to convert a zero-width wire into the identity element for whatever operation it was participating in (i.e., it becomes UInt(0) for add, UInt(1) for multiply). In the case of a Mux(), all zero-width children (including the selector), were converted into a UInt(0).

Grep the Chisel2 sources for W0Wtransform for the details. There are five distinct Node-specific implementations.

azidar, 2016-02-26:
Can anyone think of a corner case that would give unexpected behavior for Jim's approach? If not, then I can add it as an experimental feature (not formally in the spec) so we can play with it a bit and get code working.

aswaterman, 2016-02-26:
Jim's approach has worked fine in practice, but it is worth thinking this
through again in the context of a formal IR.  For example, not all binary
operators have identity elements.  It might be that the simpler and
easier-to-describe approach of treating null wires as numerical-zero in all
arithmetic is OK, too.

Here are some identities/litmus tests that I think a sane definition of
null wires would obey, but might not fall out of an obvious implementation:

  cat(null, x) == x
  not(null) == null
  add(null, null) == null (i.e., not UInt<1>(0)... right?)
  sub(null, null) == null
  sub(null, x) == -x
  add(null, x) == x
  orr(null) == 0
  xorr(null) == 0
  andr(null) == 1 (because andr(cat(x, null)) should equal
andr(cat(andr(x), andr(null))))

sdtwigg, 2016-08-15:
For the reduction operators, the IR, nominally trying to be strict, should error on reduction of zero-width things. A less-strict frontend is free to concatenate 0 (for andR and xorR) or 1 (for orR) in order to restore the always valid assumption. (It seems like an easy post width-inference peephole optimizations to remove the concatenation in the non-zerowidth case and replace with the constant in the zero-width case.)

The justification is similar to how you can't reduce an empty list but can fold on them. (One could foresee languages not wanting users to be blindly reducing empty things. This serves as a reminder to users that they properly addressed the zero-width case for their circuit when it comes up.) It also provides an alert rather than hidden surprise for the case where null is actually not zero (in case the user mistakingly remembers "null means zero" instead of "null means zero only for arithmetic operators +,-,*,/").

As an aside, you can conveniently check if a wire is zero-width via (~x === x). Correspondingly, you can't assume that value is false unless you know the widths are non-zero. This little idiom allows for you to attain different andR and orR behaviors when at zero width:
Cat(1, x).andR && (~x =/= x) will return x.andR for non-zero width and false when zero width.

jwright6323, 2016-09-13:
Is this still planned? This would be nice to have for a block I'm currently developing.

aswaterman, 2016-09-13:
Still planned, still unimplemented.

azidar, 2016-09-13:
:( its on the todo list, sorry!

azidar, 2017-02-08:
see #402

closed 2017-03-16 by azidar