chipsalliance/firrtl #66: build-scala generates java.lang.StackOverflowError with Java 1.8

ucbjrl, 2016-02-23:
I get java.lang.StackOverflowError trying to build the Scala version of firrtl with Scala 2.11, sbt 0.13.9, and (Oracle) Java 1.8.0_74 on MacOSX 10.9.5, and Scala 2.11, sbt 0.13.9, and (Oracle) Java 1.8.0_45 on Ubuntu 14.04.2 LTS.
This still occurs after deleting everything in ~/.ivy2 and ~/.m2

This doesn’t happen with Java 1.7.0_76 (on MacOSX) or 1.7.0_80 (on Ubuntu).

Here's the bottom of the Ubuntu stack dump:

    java.lang.StackOverflowError
    	at scala.tools.nsc.typechecker.Typers$Typer.checkDead(Typers.scala:111)
    	at scala.tools.nsc.typechecker.Typers$Typer.typedSelectOrSuperCall$1(Typers.scala:4811)
    	at scala.tools.nsc.typechecker.Typers$Typer.typedInAnyMode$1(Typers.scala:5343)
    	at scala.tools.nsc.typechecker.Typers$Typer.typed1(Typers.scala:5359)
    	at scala.tools.nsc.typechecker.Typers$Typer.runTyper$1(Typers.scala:5395)
    	at scala.tools.nsc.typechecker.Typers$Typer.scala$tools$nsc$typechecker$Typers$Typer$$typedInternal(Typers.scala:5422)
    	at scala.tools.nsc.typechecker.Typers$Typer.body$2(Typers.scala:5369)
    	at scala.tools.nsc.typechecker.Typers$Typer.typed(Typers.scala:5373)
    	at scala.tools.nsc.typechecker.Typers$Typer$$anonfun$99.apply(Typers.scala:4524)
    	at scala.tools.nsc.typechecker.Typers$Typer$$anonfun$99.apply(Typers.scala:4524)
    	at scala.tools.nsc.typechecker.Typers$Typer.silent(Typers.scala:680)
    	at scala.tools.nsc.typechecker.Typers$Typer.normalTypedApply$1(Typers.scala:4523)
    	at scala.tools.nsc.typechecker.Typers$Typer.typedApply$1(Typers.scala:4579)
    	at scala.tools.nsc.typechecker.Typers$Typer.typedInAnyMode$1(Typers.scala:5342)
    	at scala.tools.nsc.typechecker.Typers$Typer.typed1(Typers.scala:5359)
    	at scala.tools.nsc.typechecker.Typers$Typer.runTyper$1(Typers.scala:5395)
    	at scala.tools.nsc.typechecker.Typers$Typer.scala$tools$nsc$typechecker$Typers$Typer$$typedInternal(Typers.scala:5422)
    	at scala.tools.nsc.typechecker.Typers$Typer.body$2(Typers.scala:5369)
    	at scala.tools.nsc.typechecker.Typers$Typer.typed(Typers.scala:5373)
    	at scala.tools.nsc.typechecker.Typers$Typer.typedQualifier(Typers.scala:5471)
    	at scala.tools.nsc.typechecker.Typers$Typer.typedQualifier(Typers.scala:5477)
    	at scala.tools.nsc.typechecker.Typers$Typer.typedSelectOrSuperCall$1(Typers.scala:4811)
    	at scala.tools.nsc.typechecker.Typers$Typer.typedInAnyMode$1(Typers.scala:5343)
    	at scala.tools.nsc.typechecker.Typers$Typer.typed1(Typers.scala:5359)
    	at scala.tools.nsc.typechecker.Typers$Typer.runTyper$1(Typers.scala:5395)

azidar, 2016-02-23:
I saw this error as well - it looks like its a bug in the scala compiler (or how sbt decides what to compile). Have you tried deleting everything and starting with a fresh repo? The only (REALLY HACKY) thing that worked for me was to slowly comment out portions of the code until it correctly compiled. Then, when I uncommented the code back, it worked.

I tried deleting .ivy2 and .sbt caches, as well as sbt clean (and all things suggested here http://stackoverflow.com/questions/4483230/an-easy-way-to-get-rid-of-everything-generated-by-sbt), to no avail.

kammoh, 2016-02-25:
Hi,

Have you tried increasing the JVM stack size (-Xss)? I know this shouldn't
be really necessary but -Xss256m -Xmx4096m solved the issue for me. (bumped
to sbt 0.13.9, scala 2.11.7, running on Oracle JVM 1.8.0_74-b02)

Kamyar

ucbjrl, 2016-02-25:
-Xss256m does the trick.

aswaterman, 2016-04-15:
It appears this issue has reared its head again (link only works for people at UCB):

http://a17.millennium.berkeley.edu/jenkins/job/bare-chisel3-firrtl/50/console

aswaterman, 2016-04-15:
...time to increase the stack size to 512 MiB?

closed 2017-03-16 by azidar

abhsrivastava, 2017-11-15:
thank you so much!!!!!! -Xss256M yeah!!!!!!!!

normana400, 2018-04-19:
was the solution suggested here to add the XSS setting inside the build.sbt scalac settings or to add these settings externally?

ucbjrl, 2018-04-19:
My choice was to use [sbt-extras](https://github.com/paulp/sbt-extras), put the following inside a `$HOME/.sbtopts` file:
```bash
-no-colors
-J-Xss256m
-J-Xms1024m
```
then
```bash
export SBT_OPTS=@$HOME/.sbtopts
```
This ensures all my sbt builds use the same Xss settings.