chipsalliance/firrtl #68: scala firrtl chokes on reg ens : UInt<?>, clk with : (reset => (reset, UInt<1>("h00")))

chick, 2016-02-24:
Current master version of chisel3 test chiselTest.Counter.scala creates a firrtl file (attached) that is parsed successfully by stanza firrtl but gives the following error
```
firrtl -i *fir -o scala.vcd -X verilog
line 8:19 mismatched input '?' expecting IntLit
line 8:20 extraneous input '>' expecting {'circuit', '}', 'module', 'extmodule', 'input', 'output', 'UInt', 'SInt', 'Clock', '[', 'wire', 'reg', 'with', 'reset', 'mem', 'data-type', 'depth', 'read-latency', 'write-latency', 'read-under-write', 'reader', 'writer', 'readwriter', 'cmem', 'smem', 'inst', 'node', 'is', 'invalid', 'when', 'else', 'stop(', 'printf(', 'skip', 'infer', 'read', 'write', 'rdwr', 'old', 'new', 'undefined', 'UBits', 'SBits', '.', 'mux(', 'validif(', 'stop', 'printf', 'mux', 'validif', 'add(', 'sub(', 'mul(', 'div(', 'rem(', 'lt(', 'leq(', 'gt(', 'geq(', 'eq(', 'neq(', 'pad(', 'asUInt(', 'asSInt(', 'asClock(', 'shl(', 'shr(', 'dshl(', 'dshr(', 'cvt(', 'neg(', 'not(', 'and(', 'or(', 'xor(', 'andr(', 'orr(', 'xorr(', 'cat(', 'bits(', 'head(', 'tail(', Id}
line 8:27 no viable alternative at input 'clkwith'
line 9:4 extraneous input 'node' expecting {'}', 'module', 'extmodule'}
Exception in thread "main" firrtl.ParserException: 4 syntax error(s) detected
	at firrtl.Parser$.parse(Parser.scala:64)
	at firrtl.Driver$.compile(Driver.scala:48)
	at firrtl.Driver$.main(Driver.scala:125)
	at firrtl.Driver.main(Driver.scala)
```
with scala firrtl

[EnableTester3110576757137255563.fir.txt](https://github.com/ucb-bar/firrtl/files/143711/EnableTester3110576757137255563.fir.txt)

chick, 2016-02-24:
if you need to reproduce the firrtl file, I suggest a local edit to the end of chiselTests.Counter.scala to look like
```
class CounterSpec extends ChiselPropSpec {
  property("Counter can be en/disabled") {
    val seed = 0
    println("seed is " + seed + " X"*80)
    assertTesterPasses{ new EnableTester(seed) } }
}
```

jackkoenig, 2016-02-24:
Okay so it looks like FIRRTL Scala is failing on an unknown width:
`reg ens : UInt<?>, clk with : (reset => (reset, UInt<1>("h00")))`

I intentionally removed support for <?> a while ago in favor of just omitting a width when it is unknown. 

Adam, as far as I can tell, there is no mention of <?> in the spec, it seems that the width is optional instead. Is this correct? If so this is a chisel3 bug.

azidar, 2016-02-24:
Yeah, there isn't need for that syntax anymore, so chisel should just omit the <?>

jackkoenig, 2016-02-24:
Moved to chisel3 issues: ucb-bar/chisel3#105

closed 2016-02-24 by jackkoenig