chipsalliance/firrtl #69: Shift Right not handling shifts larger than width of argument

jackkoenig, 2016-02-24:
This problem manifests in both Scala and Stanza FIRRTL

Consider the FIRRTL:
```
circuit Bug :                                                                                                                                                                                                                                 
  module Bug :
    input in : UInt<1>
    output out : UInt<1>
    
    out <= shr(in, 1)
```
This results in the following invalid Verilog:
```Verilog
module Bug(
   input   in,                                                                                                                                                                                                                                
   output  out 
);  
   assign out = in[0:1];                                                                                               
endmodule
```
Shifting by more than the width of the operand gives similar results:
```
circuit Bug : 
  module Bug : 
    input in : UInt<5>
    output out : UInt<5>
    
    out <= shr(in, 6)     
```
```Verilog
module Bug(
   input  [4:0] in,
   output [4:0] out                                                                                                                                                                                                                           
);  
   assign out = in[4:6];                                                                                               
endmodule
```

First of all, the spec requires that the shift amount be strictly less than the width of the operand, so neither implementation is correctly implementing the spec here.

Secondly, and more importantly, since Chisel relies on FIRRTL for width-inference, it cannot prevent itself from emitting invalid FIRRTL here. Moreover, dynamic shift right does not have any constraint like this (as it obviously cannot) and just seems to punt to Verilog as in this example:
```
circuit Bug : 
  module Bug : 
    input in : UInt<5>
    output out : UInt<5>
    
    out <= dshr(in, UInt(6))                         
```
```Verilog
module Bug(                                                                                                                                                                                                                                   
   input  [4:0] in,
   output [4:0] out
);
   assign out = in >> 3'h6;
endmodule
```
Thirdly, allowing for shift amounts greater than or equal to the width of the operand makes it easier to write generators in that you don't have add special cases in your code.

I think FIRRTL should define the semantics of right shifting a number by an amount equal to or greater than the number's width to be (both are width = 1):

* 0 when unsigned
* sign bit when signed

These semantics strike me as both intuitive and necessary since they can be consistent between static and dynamic right shift.

aswaterman, 2016-02-24:
I concur with Jack's proposal for defining right shifts by amounts N >= W.  These remain consistent with the usual mathematical definitions of right shifts by N.  For unsigned, that's division by 2^N, rounding towards 0 (hence a result of 0), and for signed, that's division by 2^N, rounding towards -infinity (hence the sign bit).

palmer-dabbelt, 2016-03-06:
This is blocking my attempts to get rocket-chip ported to chisel3.

jackkoenig, 2016-03-15:
Fixed-ish 5737a8ccbf54a6d22095023205867e851e204c3f,
Complete fix comes with zero-width wires #65

closed 2016-03-15 by jackkoenig