chipsalliance/firrtl #73: Invalid handing of Primop argument to printf

jackkoenig, 2016-02-25:
Consider the following FIRRTL:
```
circuit Bug :
  module Bug :
    input clk : Clock
    input a : UInt<1>
    input b : UInt<1>

    printf(clk, UInt<1>(1), "%d\n", and(a, b))

```
Scala FIRRTL currently compiles to the following invalid Verilog:
```Verilog
module Bug(
   input   clk,
   input   a,
   input   b
);
   always @(posedge clk) begin
      `ifndef SYNTHESIS
         if(1'h1) begin
            $fwrite(32'h80000002,"%d\n",and(a, b));
         end
      `endif
   end
endmodule
```
Stanza FIRRTL does the same thing it does in #64, it puts quotes around "and(a,b)".

The spec just says the arguments to printf must have a ground type, which primops do, so I believe this should be handled by a pass. I'm not 100% on what all of the passes do but it looks like it might belong in Split Expressions or something similar. Documentation on what each pass does would be helpful.

@azidar please advise.

azidar, 2016-02-25:
Should be fixed in pull request #75.

jackkoenig, 2016-02-25:
Looks good to me!

closed 2016-02-25 by jackkoenig