chipsalliance/firrtl #74: Fix Mappers

jackkoenig, 2016-02-25:
There are a set of useful utility functions in [Utils.scala](https://github.com/ucb-bar/firrtl/blob/master/src/main/scala/firrtl/Utils.scala) that we use to apply a function to the children of a particular FIRRTL graph vertex. 

In the quick port to Scala, @azidar had to create lots of different named functions (eg. sMap, eMap...) because type erasure does not allow us to overload functions when the different parameter type is another function. For example, we can't do the following:
```Scala
def map(f: Expression => Expression, stmt: Stmt): Stmt = ...
def map(f: Stmt => Stmt, stmt: Stmt): Stmt = ...
```
Because to the JVM, both fs look the same. This was fixed for a subset before with the commented out StmtMagnet code you can still see in Utils. I am planning to move back to this style, but I have a few questions first:

1. Should we continue to call this function "map"? There obviously is similarity to the scala.collections map function. That being said, these scala.collections are *collections* whereas these FIRRTL tree vertices are not, and our semantics for map here are to apply the function to the vertex's *children*. Perhaps this is fine, or perhaps a name like *mapChildren* is more appropriate. @azidar, I remember your saying previously that LLVM uses *map* in this context, do you have any documentation to compare to? I've been looking but I don't entirely understand LLVM.
2. Perhaps this is irrelevant, but should these mappers be a method you call on FIRRTL vertices, or a function you pass the vertices to? See below. We use implicit methods because we don't want to muddy up the IR definition which is currently all case classes, and Scala does not allow you to define methods on classes unless you do so at declaration. 
```scala
def func(s: Stmt): Stmt = ...

// Function you call on Stmt vertices
def map(f: Stmt => Stmt, stmt: Stmt): Stmt = ...
// Call:
map(func, stmt)

// Implicit method on Stmt vertices
implicit class StmtMappers(stmt: Stmt) {
    def map(f: Stmt => Stmt): Stmt = ...
}
// Call:
stmt map (func)
stmt.map(func)
```

jackkoenig, 2016-02-25:
This issue may seem like it's splitting hairs, but these functions are crucial to writing passes so this API is important both internally and externally. Consider part of @grebe's pass from a while ago in FIRRTL that uses a Map[String, String] to mutate any Strings in the FIRRTL graph:
```Scala
  def renameall(s : String)(implicit map : Map[String,String]) : String =
    map getOrElse (s, s)

  def renameall(e : Expression)(implicit map : Map[String,String]) : Expression = ... 
     // similar to Stmt below

  def renameall(s : Stmt)(implicit map : Map[String,String]) : Stmt = {

    s match {
      case p : DefWire =>
        DefWire(p.info, renameall(p.name), p.tpe)
      case p: DefReg =>
        DefReg(p.info, renameall(p.name), p.tpe, p.clock, p.reset)
      case p : DefMemory =>
        DefMemory(p.info, renameall(p.name), p.seq, p.tpe, p.clock)
      case p : DefInst =>
        DefInst(p.info, renameall(p.name), renameall(p.module))
      case p : DefNode =>
        DefNode(p.info, renameall(p.name), renameall(p.value))
      case p : DefPoison =>
        DefPoison(p.info, renameall(p.name), p.tpe)
      case p : DefAccessor =>
        DefAccessor(p.info, renameall(p.name), p.dir, renameall(p.source), renameall(p.index))
      case p : OnReset =>
        OnReset(p.info, renameall(p.lhs), renameall(p.rhs))
      case p : Connect =>
        Connect(p.info, renameall(p.lhs), renameall(p.rhs))
      case p : BulkConnect =>
        BulkConnect(p.info, renameall(p.lhs), renameall(p.rhs))
      case p : When =>
        When(p.info, renameall(p.pred), renameall(p.conseq), renameall(p.alt))
      case p : Assert =>
        Assert(p.info, renameall(p.pred))
      case p : Block =>
        Block(p.stmts.map(renameall))
      case p : Stmt => p
    }
  }

```
With the appropriate map functions this pass could be rewritten without all of the calls to different overload forms of rename all like so (I think)
```Scala
  def renameallStr(s : String)(implicit m : Map[String,String]) : String =
    m getOrElse (s, s)

  def renameallExp(e : Expression)(implicit m : Map[String,String]) : Expression =  
    e map (renameallExp) map (renameallStmt)

  def renameallStmt(s : Stmt)(implicit m : Map[String,String]) : Stmt = 
    s map (renameallStmt) map (renameallExp) map (renameallStr)
```

jackkoenig, 2016-02-29:
In my opinion, one of the biggest problem with using the name "map" for these mapper functions is that it doesn't quite match the definition of map for Scala collections. For example, the definition of map on Seq is:
```scala
def map[B](f: (A) ⇒ B): Seq[B]
```
But our mappers have definitions like:
```scala
def eMap(f:Expression => Expression, stmt:Stmt) : Stmt =
```
The above definition is extremely useful in writing passes, but I think there would also be a use for a map function more similar to the Seq version above:
```scala
def map[B](f:Expression => B) : Seq[B] =
```
This map function is less useful that the above for transforming the FIRRTL graph, but is useful for extracting metadata.

azidar, 2016-02-29:
I think ideally, we want to be able to do something like

    val transformedStmt = s.map(renameStmts)

The idea of maps is it applies the given function to every element in the collection, and returns a new collection _of the same original collection type_. If we use this definition, then the current eMap definition is totally fine, because it takes a Expression (which can be a collection of Expressions) and returns the original type, Expression (which again can be a collection of Expressions).

jackkoenig, 2016-02-29:
Alright I'll get started moving it to the `stmt.map(func)` format.

We talked about this a little bit in the Chisel meeting today and there seemed to be some agreement that the name `transform` might be better here than `map` because we're transforming a tree rather than mapping on a more traditional collection. Thoughts @azidar ?

azidar, 2016-03-01:
I feel like "transform" implies transforming the node, when in reality, map actually transforms the _children_ of the node. This is much more analogous to a map applied to a collection than a transform applied to a single node.

jackkoenig, 2016-03-01:
See fix in #79 

I agree with your reasoning. I still have reservations about calling it `map`, but we can discuss another time.

jackkoenig, 2016-03-02:
Fixed in 079005f630590bdaf4671c9d8ab127b649cd61df

closed 2016-03-02 by jackkoenig