chipsalliance/firrtl #82: Scala FIRRTL Style

jackkoenig, 2016-03-03:
I think we should settle on a style for Scala FIRRTL. 

While they disagree occasionally with the official [Scala Style Guide](http://docs.scala-lang.org/style/), our AMPLab neighbors have perhaps the largest open-source Scala project: Databricks. 

The [Databricks Style Guide](https://github.com/databricks/scala-style-guide) has a lot about readability and even some good pointers for performance (when we get to that point). 

Any thoughts?

ucbjrl, 2016-03-03:
I concur with the idea of a style guide. Whatever we use for FIRRTL we should also use for chisel.

The Databricks Style Guide seems to deal with mixing Scala and Java by advising the "Java-way" over the "Scala-way" when there are conflicting (or performance) choices, most notably:

 - use Java docs, not Scala docs,
 - use Java collections over Scala ones,
 - use null instead of Option

These may be controversial for seasoned Scala programmers.

ducky64, 2016-03-03:
The null over Option sounds like an especially bad idea if we care about catching bugs at compile-time. I think some of the Chisel implementation does use nulls (explicit Some(blah) are a bit annoying to write), which sometimes creeps into unexpected places and produces cryptic, unintuitive errors.

jackkoenig, 2016-03-03:
Perhaps the Databricks guide could just be a starting point! I think we can safely ignore all of the stuff about interoperability with Java.

As for the java collections and null over Option, these suggestions are only for performance sensitive code. We're far from that point, so I think these suggestions could/should be ignored as well.

chick, 2016-09-22:
I just pushed a branch style-cleanup-take-2 that does a number of style fixes.  @jackkoenig @azidar : can you guys take a look and see if this approach to cleaning things up seems ok.  
The approach is:
Pick a type of fix, e.g. use .indices instead of (0 until a.length)
Fix all occurrences
Run test
commit

jackkoenig, 2016-09-22:
Link for the lazy https://github.com/ucb-bar/firrtl/commits/style-cleanup-take-2

I like the approach, it makes review very easy 👍

donggyukim, 2016-09-23:
Why don't we merge at leat the first five commits?

chick, 2016-09-23:
Ok with me.  @jackkoenig ? @azidar ?

donggyukim, 2016-09-23:
@chick Can you make a PR?

chick, 2016-09-23:
done

Chick Markley
Staff Programmer -- Aspire Lab
University of California, Berkeley
583-2 Soda Hall, Berkeley, CA 94720-1776
chick@berkeley.edu

azidar, 2017-03-16:
Is there a way to check the style with sbt?

chick, 2017-03-16:
@azidar Yes, [http://www.scalastyle.org/sbt.html](http://www.scalastyle.org/sbt.html) describes it and other methods.  For you though, perhaps [https://github.com/vim-syntastic/syntastic/blob/master/syntax_checkers/scala/scalastyle.vim](https://github.com/vim-syntastic/syntastic/blob/master/syntax_checkers/scala/scalastyle.vim

azidar, 2019-10-14:
Subsumed by https://github.com/freechipsproject/chisel3/pull/1126.

FIRRTL will use the same style as Chisel3.

closed 2019-10-14 by azidar