chipsalliance/firrtl #87: Supporting Verilog's "inout" construct

azidar, 2016-03-09:
There is a problem where existing Verilog boxes have ports that can be driven by different places, i.e. I2C bus. This is represented by the Verilog "inout" construct. While we don't want to directly support modeling tri-state buses, we do need a way to connect black boxed Verilog designs within FIRRTL to other black boxed Verilog designs.

As a result, we need a way to indicate that two ports are connected, but that we can make no other assumptions about it (i.e. you cannot buffer the signal, or introduce a mux, etc.)

My proposal is to add two new constructs:
1) An Analog type. An analog type has the property that no two Analog types are equivalent. Thus, the following FIRRTL causes a type error:
```
wire x : Analog
wire y : Analog
x <= y   ;ERROR: x and y do not have equivalent types
y <= x   ;ERROR: x and y do not have equivalent types
```
Another benefit is a bundle type that contains an Analog type will also fail when assigning to/from:
```
wire x : {a : Analog, b : UInt<32>}
wire y : {a : Analog, b : UInt<32>}
x <= y   ;ERROR: x and y do not have equivalent types
y <= x   ;ERROR: x and y do not have equivalent types
```
Finally, no primitive operation, mux, or node will accept an Analog-typed argument.

2) An `attach` operator. The `attach` operator takes a variable number of Analog-typed expressions, and attaches them together.
```
wire x : Analog
wire y : Analog
attach(x,y)
```
Assuming `a`, `b`, and `c` are all Analog types, the following two examples are equivalent:
```
attach(a,b)
attach(b,c)
```
and
```
attach(a,b,c)
```
Middle FIRRTL and Low FIRRTL will have the restriction that no two `attach` operators will have the same referenced expressions.

These two constructs give us the following property that Analog types will never interact with our existing connection semantics. This is for good reason - and Analog signal's direction is dynamic (or non-existent), while the connection semantics assumes a statically known direction. Rather than change our existing connection semantics and worry about how it interacts with other constructs, its better to never cause an interaction.

Thoughts?
@aswaterman @jackkoenig @jackbackrack @sdtwigg and any others?

aswaterman, 2016-03-09:
I haven't thought through the implications fully, but I think you're going to want to add the ability to attach analog types to other FIRRTL types.  Otherwise, people will just write black boxes that consist only of a single assign statement to connect them together, as a means to get around the FIRRTL limitation.  This is hokey and provides even less safety.

azidar, 2016-04-09:
Good point Andrew. Let's add an asAnalog primop that converts Clk/UInt/SInt to Analog, and add Analog as a valid input type to asUInt, asSInt, and asClock.

closed 2016-09-26 by azidar