chipsalliance/firrtl #94: Scala IR revamp

jackkoenig, 2016-03-25:
So I think there are some things that could/should change in the Scala IR definition that would clean it up a little. Some ideas are to match scala style, some are more my own stylistic opinions. 

1. Change all case class Name() to case object Name
  1. This would match scala style
1. Abbreviate/Rename various vertices 
  1. Rename Expression to Exp (we use Stmt for Statements so why not have Exp to match?)
  1. Rename DefRegister to DefReg (to match abbreviation in concrete syntax)
  1. Rename DefInstance to DefInst (to match abbreviation in concrete syntax)
  1. Rename DefMemory to DefMem (to match abbreviation in concrete syntax)
  1. Rename Conditionally to When (less important but Conditionally is kind of long)
  1. Rename Print to Printf (to match name in spec)
1. Change Primops to camel-case 
  1. ADD_OP -> AddOp (perhaps just Add?)
1. Make primops match concrete syntax name
  1. GREATER_EQ_OP -> GeqOp 
  1. Do we gain anything by making the name long?  This makes it confusing to know what names in a .fir file match the various PrimOps
  1. I'm sympathetic to the self-documenting names idea, but DYN_SHIFT_RIGHT_OP just looks ridiculous when that information can easily be put in a scala-doc
1. Use camelcase within case classes
  1. read_latency -> readLatency (or readLat)
1. Rename Direction to PortDir
1. Rename Flip to FieldDir
  1. DEFAULT is just confusing since the flip keyword really means REVERSE
  1. I think Default and Flip should be the case objects, to match "flip" keyword
1. Rename BulkConnect to PartialConnect (because spec)
1. Rename top-level trait from AST to ???
  - Something about the name bothers me. While FIRRTL circuits are ASTs, hardware is really a DAG.
  - I'd like it to be like FirrtlNode, but Node is something else, what about like FirrtlVertex?
1. Standardize use of FIRRTL vs Firrtl vs firrtl
  - Formerly, I put FIRRTL everywhere, but that doesn't really work well with camelcase
  - FIRRTLException is awkwad, I'd like FirrtlException better.

This issue may sound stupid but I'd like feedback *before* I change things so I won't change anything I shouldn't

@azidar I'd appreciate your feedback in particular

ccelio, 2016-03-25:
It's your guys's playground, but I'm somewhat allergic to short names. While it sucks for you to type more, most people are going to be reading the code. Exp - shorta reminds me of Exponents. DefInst reminds me of instructions. I'd make Stmt -> Statement to be consistent, and leave the others as they are.

The other changes, which I believe are more Scala consistent, sound good to me.

azidar, 2016-04-09:
1) Sure
2) I agree with Chris - I actually like the longer names. I'm cool with Stmt becoming Statement
3) Don't care
4) matching concrete syntax is a good idea
5) Agree
6) Sure
7) Sure
8) Definitely
9) How about "Component"? That is a term that is used heavily in the spec. Or "Element"? "IRElement"? "IRComponent?"
10) Hmm, yeah I use all three, which is bad. Since Scala is all about camel case, let's do Firrtl.

aswaterman, 2016-04-09:
re #10, Scala and Java favor all caps for acronyms in class names, e.g.
XMLLoader. So FIRRTL is more in line with that style.  Scala favors all
lowercase for acronyms as package names, e.g. firrtl.

sdtwigg, 2016-04-12:
Can you include unifying with https://github.com/ucb-bar/chisel3/blob/master/src/main/scala/Chisel/internal/firrtl/IR.scala ?

Ideally, you would have the option to avoid the overhead/potential for error of serializing chisel3 pseudo-FIRRTL output to text then deserializing to FIRRTL classes from that text.

azidar, 2016-04-12:
re 10, Chisel is technically an acronym..

aswaterman, 2016-04-12:
yeah it's also a pun on the tool though. so idk.

jackkoenig, 2016-10-18:
Resolved by #191

closed 2016-10-18 by jackkoenig