chipsalliance/firrtl #95: Name Expansion Bug

jackkoenig, 2016-03-25:
The FIRRTL below produces invalid Verilog
```scala
circuit ExpansionBug :
  module ExpansionBug :
    output io : { flip a : UInt<1>, b : UInt<1> }

    node io_a = io.a
    io.b <= io_a
```
```verilog 
module ExpansionBug(
  input   io_a,
  output  io_b
);
  wire  io_a;
  assign io_b = io_a;
  assign io_a = io_a;                                                                                                                                                                                                                                                         
endmodule
```
According to the spec section 10.1, the lowering of the io bundle should result in `io$a` and `io$b` instead of `io_a` and `io_b`

Appears to be a/the cause of https://github.com/ucb-bar/chisel3/issues/125

This bug is present in both implementations.

jackkoenig, 2016-03-25:
@azidar I remember $ being used in the past, why was this changed?

jackkoenig, 2016-03-25:
This is problematic in rocket-chip. The TestBench generates code with _ as the delineator which is correct for chisel2, but not correct with the fix in #96

aswaterman, 2016-03-26:
I think we moved to _ deliberately, for this reason among others (concerns other tools will barf; annoying for use in waveform viewers; annoying in vim).

closed 2016-03-26 by aswaterman