chipsalliance/firrtl #98: Incorrect Verilog output for extraction

aswaterman, 2016-03-31:
Out-of-range extracts don't generate an error.  That's a philosophical issue, but generating bad Verilog isn't.

I'm not sure if these are actually two separate bugs, but the following FIRRTL:

    circuit Top :
      module Top :
        input clk : Clock
        input reset : UInt<1>

        reg sterling : UInt<4>, clk
        node malory = bits(sterling, 6, 2)
        node archer = bits(sterling, 8, 5)

generates the following Verilog:

    reg [3:0] sterling;
    wire [4:0] malory;
    wire [3:0] archer;
    assign malory = sterling[6:2];
    assign archer = sterling;

The first is bad Verilog, and presumably it was supposed to be caught earlier in FIRRTL compilation.

The second omits the extraction because width(sterling) == width(archer).  This is sensible, provided the bounds were also checked earlier in FIRRTL compilation.

azidar, 2016-04-09:
So I understand you correctly, both are errors, but if FIRRTL checked it correctly, the assumption that width(sterling) == width(archer) is valid?

By the way, the CheckWidths pass is where this check should be done.

aswaterman, 2016-04-09:
I believe that is correct.

closed 2016-11-04 by aswaterman