chipsalliance/firrtl #103: Delays are not supported in Verilator

azidar, 2016-04-04:
When we initialize registers/memories with random values, we emit `#0.002;` to delay initialization, allowing the random seed to be set by the test harness (at least originally). This construct isn't supported in verilator - perhaps we should just omit it? Or should we figure out another way to set the seed properly? @aswaterman

aswaterman, 2016-04-04:
I think we still need this for VCS.  Does Verilator just ignore it, or does it issue an error?  Can we guard it with `ifndef VERILATOR or something?

aswaterman, 2016-04-04:
(it turns out `ifndef verilator, i.e. lower case, does what you'd hope)

palmer-dabbelt, 2016-04-04:
You can make Verilator ignore them, but I think we should `ifndef VERILATOR so we don't end up silently changing the behavior of other testbench code.

azidar, 2016-04-04:
Yup, `ifndef verilator works, I'll submit a pull request.

closed 2016-04-04 by aswaterman