chipsalliance/firrtl #105: Puzzling FIRRTL error

zhemao, 2016-04-05:
Got this FIRRTL error in Travis.

https://travis-ci.org/ucb-bar/rocket-chip/jobs/120801507#L1274

What does this mean, exactly. Is it an internal FIRRTL issue?

jackkoenig, 2016-04-05:
Yeah match errors should never happen, it's definitely an internal error.
The good news is I'm in the process of revamping that pass, I'll make sure
I fix whatever this is in the process.

azidar, 2016-04-06:
Yes, it is definitely a FIRRTL bug - can I see the FIRRTL source?

jackkoenig, 2016-04-06:
@azidar I'm about to PR some major changes to LowerTypes which is where the match error is, so I wouldn't look at this until then.

azidar, 2016-04-06:
The problem is a WSubAccess has leaked out of the "RemoveAccesses" passes - there shouldn't be anything wrong with the LowerTypes pass (with respect to this error), so it is probably independent of your PR. However, if you'd rather wait, that's fine with me.

jackkoenig, 2016-04-06:
Oh yeah, you're right. Nevermind, no reason to wait.

zhemao, 2016-04-06:
If you want to reproduce, checkout the axi-fix branch of RocketChip and try to build with NastiConverterTestConfig.

jackkoenig, 2016-04-06:
Here's a small example that causes the error:

This FIRRTL file can be compiled and executed as a test
```
circuit NestedIndexTester :                                                                                                                                                                                                                                                   
  module NestedIndex :
    input foo : UInt<1>[4]
    input index : UInt<2>
    output out : UInt<4>

    wire vec : UInt<4>[2]
    vec[0] <= UInt(3)
    vec[1] <= UInt(4)

    out <= vec[foo[index]]

  module NestedIndexTester :
    input clk : Clock
    input reset : UInt<1>

    inst dut of NestedIndex

    dut.foo is invalid
    dut.index <= UInt(2)

    when neq(dut.out, UInt(4)) :
      printf(clk, not(reset), "Assertion failed\nTest Failed!\n")
      stop(clk, not(reset), 1)
    else :
      stop(clk, not(reset), 0)
```

jackkoenig, 2016-04-06:
I've created a branch that has a test that will fail until this is fixed

https://github.com/ucb-bar/firrtl/tree/fix-remove-subaccess

commit 4ebc6e2d (2016-04-09): Fixed bug in Remove Accesses where a WSubAccess's index was not checked for accesses.

Fixes #105
closed 2016-04-09 by jackkoenig