chipsalliance/firrtl #106: FIRRTL disconnected wire false positive

zhemao, 2016-04-05:
https://travis-ci.org/ucb-bar/rocket-chip/jobs/120801505#L1275

The wire in question is an input which is not used internally. It should not cause an error. It is far too laborious to go in and connect every unused wire that just happens to be in a bundle.

aswaterman, 2016-04-05:
Need the FIRRTL code.  This might actually be a FIRRTL or Chisel bug, as Chisel is supposed to initialize all wires and I/Os to invalid.

zhemao, 2016-04-05:
In the FIRRTL code, the value is present in the bundle but not assigned to anywhere. So this is probably a Chisel bug.

aswaterman, 2016-04-05:
Can you provide the code?

zhemao, 2016-04-05:
I'll post the code soon.

zhemao, 2016-04-05:
Here's the file. The code in question is at 16617.

https://gist.github.com/zhemao/1a978d521229af880422b0e16b7901e6

aswaterman, 2016-04-05:
Can you test this Chisel branch? https://github.com/ucb-bar/chisel3/tree/fix-wire-not-initialized

azidar, 2016-04-05:
Why does Chisel3 always emit a DefInvalid for a Wire declaration? Shouldn't that be a bug that ideally is caught by FIRRTL?

aswaterman, 2016-04-05:
Ideally, but it's too pessimistic for writing hardware generators, where it's pragmatic to sometimes declare and not use fields.

aswaterman, 2016-04-05:
Perhaps with aggressive dead-code elimination, it would be more practical to rely on FIRRTL to catch this stuff.

azidar, 2016-04-05:
Is it a limitation of Chisel that makes it difficult to parameterize the types of declared components? Alternatively, why not expose the DefInvalid API so users that know Wires have unassigned values can be manually invalidated, instead of moving all instances of a compile-time checked error to a run-time checked error?

aswaterman, 2016-04-05:
I think it's partially a limitation of Chisel, but it's also easier to write generators in general when you don't have to explicitly parameterize everything.  This underpins my suggestion to perform DCE before performing the check.

There's an argument to be made for exposing Invalid somehow.  Always defaulting Wires to Invalid is the correct thing to do for Chisel2 compatibility, but additionally providing a safer Wire construct (akin to the difference between <= vs <-) is a good idea.

zhemao, 2016-04-05:
Well, that fix does work. But Jack is of the opinion that it removes an important type-safety feature.

azidar, 2016-04-05:
Maybe giving an option to the Wire constructor, like invalidOk = true, with the default set to invalidOk = false? Then when people are converting between Chisel 2 and 3, they now have an out, but we don't eliminate all cases of this error occurring (in the future we can use it sparingly)?

My complaints with doing DCE before the check is that dead code elimination shouldn't influence the correctness of a circuit, but instead be simply an optimization.

jackkoenig, 2016-04-05:
So I could be misreading, but I don't think we need to invalidate wires.

In this particular example we have regressIOs as a Vector of Bundles, and each member of the Vector is connected to the io bundle of different modules. ie.
```
    inst T_1126 of PutBlockMergeRegression
    T_1126.io is invalid
    T_1126.clk <= clk
    T_1126.reset <= reset

    wire regressIOs : {...}[8]
    regressIOs[0] <- T_1126.io
```
Now T_1126's io bundle does match the bundle it's being assigned to. Why is it that this assignment isn't enough? I think the Check might be wrong.

aswaterman, 2016-04-05:
I was wondering about that, too.  Notably, it doesn't complain about T_1126.

azidar, 2016-04-05:
The error says that `regressIOs[0].cache.s2_nack` specifically is not initialized. If you look at the type, this subsubfield is in the flipped direction, meaning if we expanded the assignment `regressIOs[0] <- T_1126.io`, we would get `T_1126.io.cache.s2_nack <= regressIOs[0].cache.s2_nack`. Thus, to me it appears correct.

azidar, 2016-04-05:
Also, doesn't Chisel2 error on uninitialized wires? https://github.com/ucb-bar/chisel/blob/00562dc99f80ff20f82da9a03767cee575648202/src/main/scala/hcl.scala#L98

Because it lacks DCE, FIRRTL would be more aggressive in disallowing uninitialized wires than Chisel2, but I don't think we should "throw out the baby with the bathwater" by saying Chisel3 should never error on uninitialized wires.

aswaterman, 2016-04-05:
Wires that Chisel2 DCEs are precisely the topic of this issue...

azidar, 2016-04-06:
Right. Is too much of a burden for people converting between Chisel2 and Chisel3 to set invalidOk = true for wires that Chisel2 DCE'd? Because my point is otherwise, we've actually regressed from Chisel2, because we can no longer detect an invalid circuit that Chisel2 correctly detected.

ccelio, 2016-04-06:
I'm having trouble wrapping my head around this.

Chisel2 first runs DCE, then it checks for initialized wires and errors out, correct? 

To make the Chisel3 front-end as simple as possible, the idea is to let FIRTTL handle the issue? So FIRTTL must first run DCE and then do a pass to check for initialized wires so we can error out?

aswaterman, 2016-04-06:
@azidar I'm OK with Wire() and UninitializedWire() (is that the right name?).  More flags is messy.

@ccelio @azidar's point is that he doesn't want the legality of a circuit to be defined by an optimization pass, so he doesn't like that approach.

jackkoenig, 2016-04-06:
Perhaps this isn't an either or type of question.
Perhaps the semantics of CheckInitialization can be defined to only check that "live" code is initialized. Thus you don't have to actually eliminate the code, but FIRRTL also won't error. 

Note that the code in question here isn't defined as a Wire, it's defined as a Vec of some Module IOs
```scala
 val regressIOs = Vec(regressions.map(_.io))
```

aswaterman, 2016-04-06:
@jackkoenig that makes sense if there's a definition of liveness that's suitable to bake into the language spec

jackkoenig, 2016-04-06:
It seems to me that this issue is a prime example of the delicate balancing act of what we should expect Firrtl to do, what we should expect Chisel to do, and what we should require the user to do. 

It kind of reminds me of #69 except in that case Chisel could do nothing so either Firrtl had to loosen semantics or essentially require users to abandon all benefits of width inference.

I'm not against requiring the user to explicitly invalidate things, I don't think it'll be particularly common (as evidenced by the fact that this only comes up in recent GroundTest things, not in rocket-chip itself). The big problem here is that compared to the liveness proposal, allowing/requiring the user to explicitly invalidate still has the fundamental problem that it sacrifices Firrtl's ability to provide the safety of CheckInitialization.

What if the user actually wants to create a wire with some unused fields, but with initialization checking safety on the fields that are used?

aswaterman, 2016-04-06:
@jackkoenig this issue actually comes up with great frequency, which is why chisel3 invalidates all module inputs and instance outputs.  What's showing up in groundtest amounts to a corner case that chisel3 didn't happen to be invalidating.

I'm in complete agreement with your last point (I'd like checking on used fields but not on dead fields, as is the case in chisel2).

azidar, 2016-04-06:
My complaint with the DCE is not that dead code is actually eliminated, but it is the case where a module could be either legal or illegal depending on whether some other place in my module (or even someone else's module) reads from a value or not. This property is very unintuitive for users.

As far as checking initialization on used subfields, but not unused ones, firrtl supports setting subfields to be invalid, so we would just need to expose that API.

aswaterman, 2016-04-06:
It sounds like several people would like chisel/firrtl to automatically determine that subfields are unused; exposing the Invalid construct doesn't help with that.  And it may well be possible to define liveness in an intuitive way (which probably does require that the analysis not cross module boundaries).

jackkoenig, 2016-04-06:
Perhaps FIRRTL does not require DCE but given that Chisel is intended for
writing hardware generators and requiring the user to manually specify
which fields are not connected is pretty onerous (unless we want users to
always throw the baby out with the bathwater), Chisel could require FIRRTL
to run DCE.

jackkoenig, 2016-04-06:
To clarify my previous statement, I think the case @zhemao brought up is a great example.

You can see the Chisel code in question at: https://github.com/ucb-bar/groundtest/blob/axi-fix/src/main/scala/regression.scala

We have a fairly straightforward Module interface specification
```scala

class RegressionIO(implicit val p: Parameters) extends ParameterizedBundle()(p) {
  val start = Bool(INPUT)
  val cache = new HellaCacheIO
  val mem = new ClientUncachedTileLinkIO
  val finished = Bool(OUTPUT)
}

abstract class Regression(implicit val p: Parameters)
    extends Module with HasTileLinkParameters {
  val io = new RegressionIO
}
```

And a fairly straightforward instantiation of these Modules:
```scala
class RegressionTest(implicit p: Parameters) extends GroundTest()(p) {
  disablePorts(mem = false, cache = false)

  val regressions = p(GroundTestRegressions)(p)
  val regressIOs = Vec(regressions.map(_.io))
```
with `p(GroundTestRegressions)` giving a Seq of `Module(new BlahRegression)`

There are fields within the interface that are not intended to be used. The user could of course just invalidate the whole Vec, but that is throwing out any initialization checking safety. The user could pick which fields are not being used, but that goes against much of the convenience of generators and I highly doubt anyone will actually do that. It seems to me that unless we offload some of this generator support to FIRRTL, then this is going to be a problem all over the place in Chisel. That will likely lead to Chisel just punting to run-time and invalidating everything (as evidenced by @aswaterman 's point that Chisel already invalidates module IO and instance IO already)

azidar, 2016-04-06:
Given that it appears designers would overly-use the is-invalid mechanism in FIRRTL, I am slowly being convinced. Certainly, a non-intuitive error is better than never catching it at all (which is what is happening now). In my spare time, I'll think about defining a liveness definition, and see where we can go from there. For now, the current solution is ok, so let's go ahead and merge Andrew's pull request.

jackkoenig, 2016-04-06:
Andrew's fix merged in https://github.com/ucb-bar/chisel3/commit/f9689cab3bbb5cb2cddbb429bc30d630c886034d

Should this issue be closed or the discussion left open?

azidar, 2016-04-06:
I'll start a new issue that references this thread, but more clearly states the problem/potential solution.

jackkoenig, 2016-04-06:
Sounds good!

closed 2016-04-06 by jackkoenig