chipsalliance/firrtl #108: Check for uninitialized wires is too restrictive

azidar, 2016-04-06:
As discussed previously in #106, the two semantics for illegal uninitialized wires have tradeoffs.

Proposal 1:
Any wire whose subcomponents are never assigned to is an illegal uninitialized wire.
Pro: Simple, intuitive, and locally determined semantics
Con: Difficult to write generators that correctly omit/include the wire subcomponents that are not assigned to.

Proposal 2:
Any wire whose "live" subcomponents are never assigned to is an illegal uninitialized wire.
Pro: Makes generation significantly easier (is a common case in existing Chisel code)
Con: Errors are unintuitive/non-local. Example: A new assignment to a wire can backpropagate "liveness" and cause a previously unrelated wire, that was uninitialized, to trigger an error.

The unfortunate reality is if (1) is implemented, it places too much of a burden on the designer who will likely overuse the is-invalid escape, thus negating the actually utility of (1).

Assuming we go with (2), can we come up with a intuitive definition of liveness?

aswaterman, 2016-04-06:
This is a good summary of the issue.

zhemao, 2016-04-06:
I would define "liveness" meaning that the wire satisfies one of the following conditions:

1. Combinational logic based on the wire terminates at the input of a register.
2. Combinational logic based on the wire terminates at a top-level output of the circuit.

If the wire does not satisfy either of the above, it cannot affect the proper functioning of the circuit.

aswaterman, 2016-04-06:
re: case 1: if the register doesn't propagate to an output, I don't think it should force the input to be treated as live.  In other words, registers and combinational logic need not be treated differently for the purposes of this analysis.

azidar, 2017-03-16:
This discussion is resolved - we will add the is-invalid API to Chisel.

closed 2017-03-16 by azidar