chipsalliance/firrtl #113: ExpandWhens is O(2^n)

aswaterman, 2016-04-08:
Well, it's actually O(m^n), where m is the level of when nesting.

The following short module, which has a 5-nested when statement repeated a few times, results in about 550 KB of code.  If you replicate the last 6 lines, it becomes 2.7 MB!  Holy shitballs!

https://gist.github.com/aswaterman/f354127b700643a3c7cf1758f003349a

sdtwigg, 2016-04-09:
This is really hard to solve in the general case.

You can introduce temporaries for each default to prevent yourself from inlining the default's logic more than once. Perhaps introduce a heuristic to only do this when the default is actually used more than once and is more complicated than a single node. This (even without the heuristic) basically only adds O(#when) statements to the graph AND maintains the original structure.

Trying to restructure the mux tree to only need to use a default once is tempting; however, that leads to repeating conditions....

aswaterman, 2016-04-09:
Yeah, but still, it's hard, not "hard."

sdtwigg, 2016-04-09:
Yes, true. As long as you are okay with some temporaries (which is likely preferable to duplicating conditions), finding a reasonable solution is definitely not "hard."

aswaterman, 2016-04-09:
Temps are OK.

This issue isn't an academic concern; the test case only serves to illustrate the point.  I estimate this issue overgenerates Verilog output to the tune of 50% for the stock configuration.

aswaterman, 2016-04-11:
@azidar @jackkoenig is this not a matter of flattening whens before generating muxes?  In other words, if

    y <= UInt(0)
    when a :
      when b :
        y <= c
    when d :
      when e :
        y <= f

is first transformed to
        
    y <= UInt(0)
    node t1 = and(a, b)
    when t1 :
      y <= c
    node t2 = and(d, e)
    when t2 :
      y <= f

then the current code generation scheme works fine.

sdtwigg, 2016-04-11:
That would solve your example (mainly because the when nesting in it is superfluous) but you are still susceptible to the general case where the when nesting can't be as easily removed.

You need temporaries on the **data**.

Consider:
https://gist.github.com/sdtwigg/ede0b069c4141c6dbd7af6fbc37254ea

Now, this is functionally equivalent to:
https://gist.github.com/sdtwigg/e48cbf96dc2eca35de47ef9b814db34f

However, all the y# (y0, y1, and y2) get inlined. This yields the explosion since y1 contains 3 copies of y0. y2 contains 3 copies of y1 and thus ultimately 9 copies of y0....

Note that y0 serves as the 'default' for y1 and y1 for y2....

My proposal is that, all defaults (like what y0 and y1 are) which are nontrivial (not just a simple reference or extraction of a simple reference) and used more than once are assigned to explicit temporaries to prevent this sort of explosion.

sdtwigg, 2016-04-11:
PS: If you flatten all the whens before turning into muxes ala:
```
when  A1 && B1 &&  C1 : y <= 0
when  A1 && B1 && !C1 && D1 : y <= 1
when !A1 && E1 : y <= 2

when  A2 && B1 &&  C1 : y <= 3
when  A2 && B1 && !C1 && D1 : y <= 4
when !A2 && E1 : y <= 5

when  A3 && B3 &&  C3 : y <= 6
when  A3 && B3 && !C3 && D3 : y <= 7
when !A3 && E3 : y <= 8
```

This does avoid the explosion because the flattening ensures each 'internal' default only gets used once. However, you get replication of conditions (which you could try to coalesce into temporaries as well).

You would also want to do a pass before flattening that tried to make muxes without using 'defaults' so that the obvious cases still work well
e.g. when A : y <= 0 else y <= 1

aswaterman, 2016-04-11:
I think the right solution is the one you proposed first (assign non-trivial expressions to explicit temporaries rather than inlining them).

sdtwigg, 2016-04-12:
Agreed. (I only presented the PS as a way to show how the flattening solution converges to the other solution.)

I don't think I'll have enough time in the near future to actually implement the fix since the current ExpandWhens pass is somewhat inscrutable (and i don't have a FIRRTL dev environment setup to work in). Although, while we are on the topic of quadratic expansion, for parts that are rewritten to fix it, you may want to constrain the mutability exposure of the simlist ArrayBuffer. (Especially since it is used in a two phase, only append then only read, manner.)

aswaterman, 2016-04-12:
We agree the quality of the implementation leaves something to be desired.

azidar, 2016-04-12:
> We agree the quality of the implementation leaves something to be desired.

:crying_cat_face: 

But seriously @aswaterman @sdtwigg, great analysis, thanks! @sdtwigg I'll assign it to you for now, and if someone else has time and wants to tackle it, feel free.

aswaterman, 2016-04-12:
I just mean the straight port to scala is rough around the edges.
It's getting better.

colinschmidt, 2016-04-21:
When running Hwacha through firrtl the expand whens pass took more than 30 minutes and then crashed due to running out of heap space (I had given java 32 gigabytes)

```
[INFO] Starting Expand Whens
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
        at firrtl.Mappers$ExpMagnet$.forExp(Mappers.scala:103)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$2.apply(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$2.apply(Passes.scala:1076)
        at firrtl.Mappers$ExpMap.map(Mappers.scala:144)
        at firrtl.passes.ExpandWhens$.firrtl$passes$ExpandWhens$$replace_void$1(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$1.apply(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$1.apply(Passes.scala:1076)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.immutable.List.foreach(List.scala:381)
        at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
        at scala.collection.immutable.List.map(List.scala:285)
        at firrtl.Mappers$ExpMagnet$$anon$5.map(Mappers.scala:111)
        at firrtl.Mappers$ExpMap.map(Mappers.scala:144)
        at firrtl.passes.ExpandWhens$.firrtl$passes$ExpandWhens$$replace_void$1(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$1.apply(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$1.apply(Passes.scala:1076)
        at firrtl.Mappers$ExpMagnet$$anon$5.map(Mappers.scala:109)
        at firrtl.Mappers$ExpMap.map(Mappers.scala:144)
        at firrtl.passes.ExpandWhens$.firrtl$passes$ExpandWhens$$replace_void$1(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$1.apply(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$1.apply(Passes.scala:1076)
        at firrtl.Mappers$ExpMagnet$$anon$5.map(Mappers.scala:109)
        at firrtl.Mappers$ExpMap.map(Mappers.scala:144)
        at firrtl.passes.ExpandWhens$.firrtl$passes$ExpandWhens$$replace_void$1(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$1.apply(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$1.apply(Passes.scala:1076)
        at firrtl.Mappers$ExpMagnet$$anon$5.map(Mappers.scala:109)
        at firrtl.Mappers$ExpMap.map(Mappers.scala:144)
        at firrtl.passes.ExpandWhens$.firrtl$passes$ExpandWhens$$replace_void$1(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$1.apply(Passes.scala:1076)
        at firrtl.passes.ExpandWhens$$anonfun$firrtl$passes$ExpandWhens$$replace_void$1$1.apply(Passes.scala:1076)
```

sdtwigg, 2016-04-21:
What module are you elaborating when this shows up?
Can manually apply the fix until someone has time to do the proper workaround.

commit 0f1cedd7 (2016-04-26): Rewrite ExpandWhens to memoize complex default values

Fixes #113
commit 935ec9c2 (2016-04-28): Rewrite ExpandWhens to memoize complex default values

Fixes #113 and Fixes #150
commit f0034626 (2016-04-29): Rewrite ExpandWhens to memoize complex default values

Fixes #113 and Fixes #150
commit 40727263 (2016-05-03): Rewrite ExpandWhens to memoize complex default values

Fixes #113 and Fixes #150
closed 2016-05-04 by jackkoenig