chipsalliance/firrtl #131: Chirrtl port clocks get used before their declaration

jackkoenig, 2016-04-19:
The following Chirrtl overwrites the clock for a port:
```
circuit Bug :
  module Bug :
    input clk : Clock
    smem ram : UInt<32>[128]

    node newClock = clk
    infer mport y = ram[UInt(2)], newClock
    y <= UInt(3)
```
Running the passes to remove Chirrtl results in:
```
circuit Bug :                                                                                                                                                                                                                                                                
  module Bug : 
    input clk : Clock
    mem ram : 
      data-type => UInt<32>
      depth => 128
      read-latency => 1
      write-latency => 1
      writer => y
      read-under-write => undefined
    ram.y.addr is invalid
    ram.y.clk <= newClock
    ram.y.en <= UInt<1>("h0")
    ram.y.data is invalid
    ram.y.mask <= UInt<1>("h0")
    node newClock = clk
    ram.y.addr <= UInt<2>("h2")
    ram.y.en <= UInt<1>("h1")
    ram.y.data <= UInt<2>("h3")
    ram.y.mask <= UInt<1>("h1")
```
newClock is being used before it is declared

jackkoenig, 2016-04-19:
I have modified RemoveCHIRRTL to assign to a port clock at mport declaration rather than at the memory definition. Now the problem is for memory ports defined in when statements. The assignment of the port clock needs to be done unconditionally, so I need to extract this to the top-level.

Do y'all think it would be best to do this in ExpandWhens and treat assignment to port clocks as special cases? Or perhaps I could just assign all port clocks at the very end of the module (ugly but could be done in RemoveCHIRRTL)

jackkoenig, 2016-04-20:
This is actually a fairly complicated problem to solve correctly. The semantics of defining a chirrtl port in a conditionally statement should be no different than defining a register, which means it should essentially be pulled out the conditional block. This leads to the obvious problem that anything it depends on may still be guarded by the condition, essentially requiring us reimplement expand whens.

jackkoenig, 2016-04-20:
@azidar Any thoughts on this?

azidar, 2016-04-20:
Yeah, so at the DefMemory location assign `ram.y.clk is invalid`, and assign the original clock to it at the CHIRRTL memory port location. Expand When's will create a ValidIf, but the ValidIf will get removed as an optimization. It is a bit of a hack, but I think a ValidIf on clocks is probably really special anyways.

jackkoenig, 2016-04-20:
Yeah that's what I have in my local "fix", it just seemed undesirably hacky, I'll add some tests and PR it

azidar, 2016-04-20:
Yeah :( This is really the only blemish between CHIRRTL and FIRRTL, which was a necessary tradeoff to give Chisel3 the desired expressiblity. I imagine once the Chisel3 FIRRTL IR is the same as the real FIRRTL IR, we could potentially do a more robust fix.

jackkoenig, 2016-04-20:
Well, I think dealing with this wart and making sure it's documented is probably vastly better than trying to redo expand whens in Remove CHIRRTL!

commit 13cc1589 (2016-04-21): Change RemoveCHIRRTL to define port clocks at CHIRRTL port definition

Invalidate clock at mem definition. Fixes #131
closed 2016-04-21 by aswaterman