chipsalliance/firrtl #132: Recursive Instantiation of Modules allowed

jackkoenig, 2016-04-20:
Recursive instantiation of modules is allowed, this obviously doesn't make sense.
```
circuit Foo : 
  module Foo : 
    input a : UInt<32>
    output b : UInt<32>
    inst bar of Bar
    bar.a <= a
    b <= bar.b
  
  module Bar : 
    input a : UInt<32>
    output b : UInt<32>
    inst foo of Foo
    foo.a <= a
    b <= foo.b
```

Results in the Verilog:
```
module Foo(                                                                                                            
  input  [31:0] a,                                                                                                     
  output [31:0] b                                                                                                      
);                                                                                                                     
  wire [31:0] bar_a;                                                                                                   
  wire [31:0] bar_b;                                                                                                   
  Bar bar (                                                                                                            
    .a(bar_a),                                                                                                                                                                                                                                
    .b(bar_b)                                                                                                          
  );                                                                                                                   
  assign b = bar_b;                                                                                                    
  assign bar_a = a;                                                                                                    
endmodule                                                                                                              
module Bar(                                                                                                            
  input  [31:0] a,                                                                                                     
  output [31:0] b                                                                                                      
);                                                                                                                     
  wire [31:0] foo_a;                                                                                                   
  wire [31:0] foo_b;                                                                                                   
  Foo foo (                                                                                                            
    .a(foo_a),                                                                                                         
    .b(foo_b)                                                                                                          
  );                                                                                                                   
  assign b = foo_b;                                                                                                    
  assign foo_a = a;                                                                                                    
endmodule             
```

azidar, 2016-04-21:
Yup. This should get checked in CheckHighForm.

chick, 2016-06-09:
Working on it

jackkoenig, 2016-10-18:
Fixed by #201

closed 2016-10-18 by jackkoenig