chipsalliance/firrtl #143: Name clashes (conflicts with unfortunately chosen user names)

ucbjrl, 2016-04-22:
In the process of porting Palmer's torture tester to Chisel3, I ran into the following:

    class NameClash extends Module {
      val io = new Bundle {
        val in0 = Bits(INPUT, width=3);
        val in1 = Bits(INPUT, width=64);
      }
      val io_in0 = Wire(Bits(width = 3))
      io_in0 := io.in0;
    }

generates the following FIRRTL:

    module NameClash : 
      input clk : Clock
      input reset : UInt<1>
      output io : {flip in0 : UInt<3>, flip in1 : UInt<64>}
    
      io is invalid
      wire io_in0 : UInt<3>
      io_in0 is invalid
      io_in0 <= io.in0

which generates the following verilog:

    module NameClash(
      input   clk,
      input   reset,
      input  [2:0] io_in0,
      input  [63:0] io_in1
    );
      wire [2:0] io_in0;
      assign io_in0 = io_in0;
    endmodule

which in turn provokes the following from verilator:

    %Error-ASSIGNIN: .../NameClashTester.v:8: Assigning to input variable: io_in0

jackkoenig, 2016-04-22:
#120 fixes this issue, I guess it's about time that gets merged. It has some conflicts so I'll resolve those here in a bit and then get it merged

ucbjrl, 2016-04-22:
Resolved by #120

closed 2016-04-22 by ucbjrl