chipsalliance/firrtl #145: Nested PrimOps not supported (without SplitExp)

jackkoenig, 2016-04-26:
If you run FIRRTL on the following code without SplitExpressions you get an exception
```
circuit Test :
  module Test : 
    input a : UInt<1>
    input b : UInt<1>
    input c : UInt<1>
    output z : UInt
    z <= and(and(a,b),c)
```
`[error] (run-main-b) firrtl.EmitterException: Can't emit firrtl.DoPrim as PrimOp argument`
Should this be supported or is it correct that SplitExpressions is required for correctness?

azidar, 2016-04-26:
IMO, until we can guarantee that the emitted Verilog is correct, it should error.

aswaterman, 2016-04-26:
I agree.  Right now it's definitely not correct (e.g., bits(mux()) was being emitted as mux(), silently dropping the bits()).

In general, it's difficult to emit correct Verilog with nested expressions, because of the strange width and sign rules.

jackkoenig, 2016-04-27:
Makes sense to me!

closed 2016-04-27 by jackkoenig