chipsalliance/firrtl #151: Q: What is stanza?

sirinath, 2016-04-29:
What is stanza? Can you direct me to a website or documentation?

CuppoJava, 2016-04-29:
Hi Suminda. 

Stanza is a new general purpose programming language being designed in our lab. It's designed specifically to be good at architecting complex programs, and to be productive across the whole software development timeline, from prototyping through production. We chose to design and develop the original FIRRTL implementation in Stanza because it offered us the most productive framework out of all the existing languages, and it has served us well.

You can visit lbstanza.org for more information. That's a very old website, and we are preparing a public release within this month.

  -Patrick

sirinath, 2016-04-29:
Thanks. Is the language open sourced?

CuppoJava, 2016-04-29:
Yes. It has a 3-clause BSD license.

sirinath, 2016-04-29:
Where is the repo?

CuppoJava, 2016-04-29:
The repo is not public yet. We are preparing a public release this month.
But all the source files are available in the download package on the
website.

Cheers.
  -Patrick

closed 2016-05-03 by azidar

sirinath, 2016-05-12:
Can you ping me when the repo becomes available?

CuppoJava, 2016-05-12:
Sure!