chipsalliance/firrtl #156: Poor QoR for register updates

aswaterman, 2016-05-03:
The expand-whens and split-expressions fixes, while very much necessary, result in a QoR regression.  Gated register updates are now emitted as a mux rather than an if-statement.  Sometimes, this prevents Design Compiler from inferring a clock gate.  Other times, it precludes unrelated optimizations because it fails to grasp the high-level intent of the code.

Basically, FIRRTL code like

    reg foo : UInt<64>, clk
    when bar :
      foo <= baz

now emits Verilog code like

    assign tmp = bar ? baz : foo;
    always @(posedge clk)
      foo <= tmp;

when ideally it would emit something more like

    always @(posedge clk)
      if (bar)
        foo <= baz;

azidar, 2016-05-03:
I think with the new SplitExpressions pass in #146, we can stop splitting on Mux expressions and it will work with the previous problems, as well as fix this QOR issue.

aswaterman, 2016-05-03:
@azidar that makes sense, but I quickly tried removing Mux from SplitExp and doing so had no effect.  Seems like this is above my pay grade

azidar, 2016-05-03:
Whoops, I referenced the wrong PR - I meant in #149. Let me submit a PR based off of that branch, and you can try it out.

azidar, 2016-05-03:
Actually, it isn't splitting muxes as is, do you see a problem with #149?

aswaterman, 2016-05-03:
#149 doesn't have the problem, but it has the code size explosion.  The combination of #146 and #149 still has the QoR problem.

azidar, 2016-05-03:
Ah ok, I think I understand the problem. After expand whens, we need a pass that inlines nodes that have temporary names and are only read from once. I'll make a PR.

aswaterman, 2016-05-03:
Ah, that sounds right.

azidar, 2017-03-16:
Is this still a problem?

aswaterman, 2017-03-16:
I spot-checked the Rocket pipeline using the latest Firrtl, and it looks good; feel free to close.

closed 2017-04-26 by jackkoenig