chipsalliance/firrtl #159: Threading issues

jackkoenig, 2016-05-04:
As the number of tests grows, thread unsafe behavior creeps out of the woodwork. I have no idea what is causing it but something is breaking parallel running of tests. I have to babysit Travis to restart it if it fails randomly, and locally I usually have to run `sbt test` followed by `sbt test-quick` to re-run anything that fails to make sure it isn't actually failing.

kammoh, 2016-05-04:
Would it possibly be the case that any of these failing tests are verilator runs? There's a path collision problem in chisel3 verilator invokation that I can PR my fix.

jackkoenig, 2016-05-04:
The fails that I'm seeing are actually errors during FIRRTL execution.

Is the problem you're referring to the one where if you have a verilog file in the chisel3 (or firrtl) directory with the same name as one in a /tmp/ directory that you're trying to build, Verilator will grab the wrong one?

donggyukim, 2016-08-01:
It seem like firrtl is thread unsafe again.

donggyukim, 2016-08-01:
@jackkoenig "Random invalid strings should fail" in StringSpec keeps failing on travis. It may also affect threading tests?

jackkoenig, 2016-08-01:
I have PRed a fix to StringSpec. That shouldn't have anything to do with threading though. If the Threading tests are failing, something is most likely not threadsafe.

jackkoenig, 2016-08-02:
I believe the issue is in 94b28438d1658d0835122b8c27bbbf3753892475

I haven't figured out what exactly in the code is doing it, but it seems the previous commit does not fail the MultiThreading test.

jackkoenig, 2016-08-02:
I'm pretty sure the issue is that the instance loop checking uses mname which is a global variable

ucbjrl, 2019-03-01:
Running CI tests under Jenkins has revealed another concurrency issue. We find cases where tests are caught in a tight loop in:
```scala
/* inlined HashTable.findEntry0 to preserve its visibility */
private[this] def findEntry(key: A, h: Int): Entry = {
  var e = table(h).asInstanceOf[Entry]
  while (notFound(key, e))
    e = e.next
  e
}
```
due to the fact that `e.next` point to `e`.

This has been traced to the following code in `src/main/scala/firrtl/passes/RemoveAccesses.scala`:
```scala
// This improves the performance of this pass
private val createExpsCache = mutable.HashMap[Expression, Seq[Expression]]()
private def create_exps(e: Expression) =
  createExpsCache getOrElseUpdate (e, firrtl.Utils.create_exps(e))
```
This cache needs to be protected against simultaneous access from multiple threads and the problem can be remedied by:
 - using a per-circuit cache,
 - converting RemoveAccess from an object to a class and making it scoped per transform instance.

ucbjrl, 2019-03-13:
I'll close this and open a new issue - we should have new issues for each threading issue we find and correct.

PR #183 (2016-06-07): Fix non-thread safe Serialize by splitting it into class and object
As far as I can tell this fixes the non-thread safe behavior we've been seeing.

It is kind of awkward that the BigInt serialize function is part of Serialize since this function is really for nodes of the FIRRTL AST. You can see in Checks.scala that calling it requires (new Serialize).serialize which is kind of awkward, perhaps this function should be replaced with a Utils function called prettyPrint or something?

Resolves #181 
Fixes #159 ?
PR #214 (2016-08-02): Fix use of global state in instance loop checking
Also increase sensitivity of thread safety checking
Fixes #159
commit 20361627 (2016-08-02): Fix use of global state in instance loop checking

Also increase sensitivity of thread safety checking
Fixes #159
closed 2019-03-13 by ucbjrl