chipsalliance/firrtl #161: Modules with Unknown IO Widths Inferred Incorrectly

jackkoenig, 2016-05-09:
Related to the discussion in ucb-bar/chisel3#142

If you have a module with unspecified widths and instantiate it such that the different instantiations should result in different widths, the module is inferred to have the max of the two widths. This is functionally correct but is wasteful in the hardware generated. Consider the example below.
```
circuit Test :
  module WidthUnknown :
    input a : UInt
    input b : UInt
    output z : UInt
    z <= add(a, b)

  module Test :
    input a : UInt<16>
    input b : UInt<16>
    input c : UInt<32>                                                                                                                                                                                                                        
    input d : UInt<32>
    output x : UInt
    output y : UInt

    inst foo of WidthUnknown
    foo.a <= a
    foo.b <= b
    x <= foo.z

    inst bar of WidthUnknown
    bar.a <= c
    bar.b <= d
    y <= bar.z
```
Which results in:
```
circuit Test : 
  module WidthUnknown : 
    input a : UInt<32>
    input b : UInt<32>
    output z : UInt<33>
    z <= add(a, b)
  
  module Test : 
    input a : UInt<16>
    input b : UInt<16>
    input c : UInt<32>
    input d : UInt<32>
    output x : UInt<33>
    output y : UInt<33>
    inst foo of WidthUnknown
    inst bar of WidthUnknown
    foo.a <= a
    foo.b <= b
    bar.a <= c
    bar.b <= d
    x <= foo.z
    y <= bar.z
```

azidar, 2016-05-09:
This is the intended behavior (it isn't a bug). As to whether this behavior should be changed, I don't see any alternative solutions other than duplicating the module and inferring separate widths. WidthUnknown's ports need to be inferred to something, and the only legal value (if we don't duplicate the module) is the max of the two widths.

jackkoenig, 2016-05-09:
I suppose saying "Inferred Incorrectly" was itself incorrect since this is the intended behavior.

That being said, I think this behavior kind of undermines the point of inferring the widths on modules. My understanding is that the main argument is for code reuse, but in seems that doing so with the current behavior will result in unnecessary hardware. I think in the case where a module has unspecified widths in makes perfect sense to duplicate the module, infer separately, and then recombine if possible (ie. the inferred widths match).

aswaterman, 2016-05-09:
In any case, it doesn't match the expected behavior for Chisel, where two instances of the same module shouldn't be coupled in their implementation.  We'd have to emit different code for each instance of a module in the current FIRRTL regime, which seems to me like the wrong approach.  I'd advocate for changing the spec and duplicating modules with inferred input widths prior to width inference.

azidar, 2016-05-10:
I don't know if I agree - when I think of instantiating two instances of the same module, I think its pretty intuitive that the instances should be identical. That being said, Chisel has to emit separate modules anyways for each instance because there isn't any guarantee that the module constructor, when called twice, will emit identical modules, so this really isn't an undue burden on Chisel. Similarly, there can certainly be a Firrtl pass that merges Modules if they happen to be identical. That being said, I'm not diametrically opposed to the idea of changing the spec, and you make a good point. I think it has more to do with what is Firrtl - does it allow for some generator-like behavior (i.e. generating different modules for different widths) or is it more the straight RTL structural representation (i.e. modules are more like black boxes that you stamp out and thus coupling the implementations).

aswaterman, 2016-05-10:
I do see what you mean.

I'd still advocate for changing the spec.  The coupled implementations don't, in my opinion, follow the principle of least surprise.  You might make some subsystem that instantiates a module with inferred input widths, then down the road, instantiate that same module in an unrelated subsystem--and in so doing change the behavior of the first subsystem.  This also precludes unit-testing the individual subsystems, as the generated hardware for one differs in the absence of the other.

It's tangential, but I remain concerned about global width inference becoming a performance bottleneck.  Of course, the implementation can always take advantage of modules that happen to have fully-specified input widths and solve a local problem opportunistically.  It's just a little extra complexity.

jackkoenig, 2017-02-06:
Solved similarly to https://github.com/ucb-bar/chisel3/issues/142

closed 2017-02-06 by jackkoenig