chipsalliance/firrtl #173: SInts not respected in Legalize for connection where rhs is wider SInt than lhs

jackkoenig, 2016-05-17:
Similarl to #169
```
 circuit Test :
  module Test :
    input b : SInt<32>
    output x : SInt<16>
    skip
    x <= b
```
->
```
 circuit Test :
  module Test :
    input b : SInt<32>
    output x : SInt<16>
    skip
    x <= bits(b, 15, 0)
```

jackkoenig, 2016-05-17:
Fix coming for this as well

jackkoenig, 2016-05-18:
Per my proposed fix, according to the spec assignment of wider ground types to narrower ground types is not even allowed. Is this changing @azidar or do we need to fix the compiler?

azidar, 2016-05-24:
Yeah, so first we should change Chisel3 to always emit partial connects, and then we can check for this in checkwidths

aswaterman, 2016-05-24:
I tried, but started to get match errors in Remove CHIRRTL:

    Exception in thread "main" scala.MatchError: (UIntType(IntWidth(1)),UnknownType()) (of class scala.Tuple2)
      at firrtl.Utils$.get_valid_points(Utils.scala:311)
      at firrtl.passes.RemoveCHIRRTL$.firrtl$passes$RemoveCHIRRTL$$remove_chirrtl_s$1(Passes.scala:1397)
      at firrtl.passes.RemoveCHIRRTL$$anonfun$firrtl$passes$RemoveCHIRRTL$$remove_chirrtl_s$1$3.apply(Passes.scala:1411)

aswaterman, 2016-05-24:
The chisel3 branch is here: https://github.com/ucb-bar/chisel3/tree/partial-connect

commit 4c4b64dc (2016-09-12): Change Legalize Connect to respect SInt

Legalize will wrap the rhs of a connect statement with a bit select primop if
the lhs is of smaller width than the rhs. This bit select is now wrapped in a
asSInt cast if the original rhs was an SInt so that is has the correct type.

Fixes #173
commit b70a6688 (2016-09-12): Change Legalize Connect to respect SInt

Legalize will wrap the rhs of a connect statement with a bit select primop if
the lhs is of smaller width than the rhs. This bit select is now wrapped in a
asSInt cast if the original rhs was an SInt so that is has the correct type.

Fixes #173
commit b658246e (2016-09-12): Change Legalize Connect to respect SInt

Legalize will wrap the rhs of a connect statement with a bit select primop if
the lhs is of smaller width than the rhs. This bit select is now wrapped in a
asSInt cast if the original rhs was an SInt so that is has the correct type.

Fixes #173
commit 5bdecc1d (2016-09-13): Change Legalize Connect to respect SInt

Legalize will wrap the rhs of a connect statement with a bit select primop if
the lhs is of smaller width than the rhs. This bit select is now wrapped in a
asSInt cast if the original rhs was an SInt so that is has the correct type.

Fixes #173
closed 2016-09-13 by jackkoenig