chipsalliance/firrtl #192: Conditionals Syntactic Shorthands

kammoh, 2016-06-13:
Current FIRRTL spec mentions a "syntactical shorthand" for else-when chains (in 5.9.1: Syntactic Shorthands) which is not supported by the current Scala implementation (I guess some of the other shorthands are not supported either). I was wondering if this is something which is planned to be implemented in the future or if it's probable that the spec is going to drop this altogether.

azidar, 2016-06-13:
@jackkoenig are there plans to support this in the Scala implementation?

jackkoenig, 2016-06-13:
So it turns out that the shorthand for else-when chains is difficult to support in the current implementation of the parser. They were supported until recently, but I dropped them with the addition of source locators. The problem how do we correctly deal with source locators on shorthand when statements? ie.

```
when condition : @[when source locator]
    some <= connection @[connect source locator]
```

So how do we do this on shorthand whens?
```
; Like so?
when condition : @[when source locator] (some <= connection @[connect source locator])

; Or perhaps
when condition : (some <= connection @[connect source locator]) @[when source locator]
```

The main issue is not with the ANTLR-generated parser but with how this is handled in the preprocess step (ie. Translator). I do not have plans to add support in the short term because the Translator has been a real time sink. In my opinion, one of the following 3 things should happen:  

1. Translator is replaced by a 2nd ANTLR grammar that does the preprocess step
1. The primary ANTLR grammar handles white space scoping and Translator can be deleted
1. White space scoping is abandoned in favor of brackets

In any of the above cases, these syntactical shorthands should be relatively easy to support although it is a reasonable question why shorthand is needed in an IR.

For the record, register shorthand is supported because there's only 1 source locator so there was no ambiguity to deal with.

azidar, 2016-06-15:
Sounds like the plan is to support it eventually :)

closed 2016-06-15 by azidar

kammoh, 2016-06-20:
Hi,
Just made a PR (#194) regarding this issue, adapting @jackkoenig 's solution "2", handling while-space scoping in the lexer. I've also made some clean-up to the parser code, which showed up to be quite brittle.