chipsalliance/firrtl #193: Width inference incorrect for subtraction of SInt by UInt

chick, 2016-06-16:
Consider
```
-4.SInt<3> -  5.UInt<3> = -9.SInt<5>
   100     -     101    =  11001
   -4      -       5    =     -9
```  
The Firrtl Spec incorrectly states that ```SInt<x> - UInt<y>  => SInt<max(x,y)+1>``` in 7.2 page 40.

aswaterman, 2016-06-16:
Should be max(x+1,y+2), right?

chick, 2016-06-16:
The same width error is true of UInt - SInt  should be ```max(w1, w2) + 2```

chick, 2016-06-16:
I think your are right, but making a test to convince myself

chick, 2016-06-16:
@aswaterman it looks to me like your rule is correct except when the SInt width is 1, e.g.
```i.S<1> + j.U<y>```  in which case the width is ```y + 1```. 
Example: ```-1.S<1> - 8.U<4> => -9.S<5>```  though predicted ```max(x+1, y+2) == 6```
Any problems with this?  I assume we want Firrtl behavior to be precise.

aswaterman, 2016-06-16:
Yeah, though it is asymmetric.  That is, S<1> - U<n> needs only n+1 bits, but U<n> - S<1> needs the full n+2.

I think taking advantage of the special case makes sense.

chick, 2016-06-16:
Yes, The ```j.U<y> - i.S<x>`` follows the rule even when x is 1
Verified both for Widths from 1 to 100

commit 24ffde94 (2016-09-22): Fixed width inference for add, sub (#312)

Fixes #308 
Fixes #193
closed 2016-09-22 by jackkoenig