chipsalliance/firrtl #196: Tail operation

da-steve101, 2016-07-11:
Shouldn't tail indicate how many bits from the end to keep rather than drop?
Currently this returns myTail = 42 ( taking last 7 bits) rather than myTail = 2 ( taking last 3 bits )
The only other language I can think of that uses the tail(3) format instead of drop(7) (more the behaviour demonstrated here) is [R](http://rfunction.com/archives/699).
```scala
class TailTest extends Module {
  val io = new Bundle {
    val a = UInt( INPUT, 10 )
    val b = UInt( OUTPUT, 10 )
  }
  io.b := io.a.tail(3)
}

class TailTester extends BasicTester {
  val dut = Module( new TailTest )
  dut.io.a := UInt( 682, 10 )
  assert( dut.io.b === UInt( 2, 10 ) )
  stop()
}

class TailOpsSpec extends ChiselPropSpec with Matchers {
  property("TailTester should return the correct result") {
    assertTesterPasses{ new TailTester }
  }
}
```
See the .fir and .v file in [TailTester.zip](https://github.com/ucb-bar/firrtl/files/356474/TailTester.zip)

azidar, 2016-07-27:
Great point - tail is definitely a confusing op, (at the time I couldn't think of a better name). The original reasoning was the head/tail functions on lists in Scala. I do really like `drop`, potentially we can change the name in a future version. I'll add it to the future-release.txt file in the spec folder.

azidar, 2016-07-27:
#205

closed 2016-07-27 by azidar