chipsalliance/firrtl #200: Infinite loop (?) in Infer Widths pass.

ccelio, 2016-07-22:
As far as I can tell, I'm experiencing an infinite loop in the Infer Widths pass. It runs for ~3 minutes and then crashes the GC due to an OOM error.

There is no way obvious way for me to diagnosis this or figure out where in my Chisel design this error originates from.

````
[INFO] Starting Infer Widths
Exception in thread "main" java.lang.OutOfMemoryError: GC overhead limit exceeded
	at firrtl.Utils$.firrtl$Utils$$apply_t$1(Utils.scala:672)
	at firrtl.Utils$$anonfun$firrtl$Utils$$apply_t$1$1.apply(Utils.scala:672)
	at firrtl.Utils$$anonfun$firrtl$Utils$$apply_t$1$1.apply(Utils.scala:672)
	at firrtl.Mappers$TypeMagnet$$anon$8$$anonfun$map$1.apply(Mappers.scala:155)
	at firrtl.Mappers$TypeMagnet$$anon$8$$anonfun$map$1.apply(Mappers.scala:155)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.Mappers$TypeMagnet$$anon$8.map(Mappers.scala:155)
	at firrtl.Mappers$TypeMap.map(Mappers.scala:172)
	at firrtl.Utils$.firrtl$Utils$$apply_t$1(Utils.scala:672)
	at firrtl.Utils$$anonfun$firrtl$Utils$$apply_t$1$1.apply(Utils.scala:672)
	at firrtl.Utils$$anonfun$firrtl$Utils$$apply_t$1$1.apply(Utils.scala:672)
	at firrtl.Mappers$TypeMagnet$$anon$8$$anonfun$map$1.apply(Mappers.scala:155)
	at firrtl.Mappers$TypeMagnet$$anon$8$$anonfun$map$1.apply(Mappers.scala:155)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.Mappers$TypeMagnet$$anon$8.map(Mappers.scala:155)
	at firrtl.Mappers$TypeMap.map(Mappers.scala:172)
	at firrtl.Utils$.firrtl$Utils$$apply_t$1(Utils.scala:672)
	at firrtl.Utils$$anonfun$firrtl$Utils$$apply_t$1$1.apply(Utils.scala:672)
	at firrtl.Utils$$anonfun$firrtl$Utils$$apply_t$1$1.apply(Utils.scala:672)
	at firrtl.Mappers$TypeMagnet$$anon$8$$anonfun$map$1.apply(Mappers.scala:155)
	at firrtl.Mappers$TypeMagnet$$anon$8$$anonfun$map$1.apply(Mappers.scala:155)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
make: *** [/scratch/celio/boom-chip/emulator/generated-src/Top.BOOML1Config.v] Error 1
````

jackkoenig, 2016-07-22:
I'm not that familiar with the implementation of Infer Widths, but my guess is that something in there is struggling with the sheer size of BOOM.

There could, of course, be some implementation mistake that once fixed will remedy this problem, but it does look like another motivating example for @sdtwigg's suggestion that we don't allow unknown widths on module boundaries, and thus infer widths can be done locally on each module thus likely saving a lot of time and memory. 

See previous discussion:  https://github.com/ucb-bar/chisel3/issues/142

ccelio, 2016-07-22:
I've tended to always specify my widths explicitly, but I'll check to see if any module boundaries are underspecified.

jackkoenig, 2016-07-22:
My understanding is that the current implementation always treats it as a global problem whether or not it needs to be, perhaps a stopgap solution for BOOM would be to change Infer Widths to act locally when it can and globally only when it has to. Thoughts @azidar?

azidar, 2016-07-22:
I doubt this is actually the problem, but I'm looking into it. @ccelio, could you email me the BOOM.fir file that triggers this error?

ccelio, 2016-07-22:
@azidar I put it in /nscratch/celio/for_adam/.

azidar, 2016-07-26:
From what I can tell, the problem stems from the number of Firrtl objects in BOOM after the Expand Whens pass (when the circuit is in Middle Firrtl Form). At this point in time in the compiler, all connections/whens/etc. have been expanded. However, the Firrtl types have not been lowered. This means that simple connections have now expanded into a much larger number of statements, and those statements contain even more expressions to index the aggregate type. Finally, those expressions each have a type, often represented by a huge bundle type. Below is a simple circuit illustrating the issue.

To estimate the memory footprint, we will count the number of strings in the in-memory Firrtl representation of each statement. To start, we have two Bundle-typed wires, and we connect them (High Firrtl).
```
wire a: {x: {i: UInt, j: UInt}, y: {i: UInt, j: UInt}} //7
wire b: {x: {i: UInt, j: UInt}, y: {i: UInt, j: UInt}} //7
a <= b //7 + 7 (remember each reference has a type which contains strings)
```
totaling 28 strings. In the point of the compiler after the ExpandWhens pass (Middle Firrtl), we see:
```
wire a: {x: {i: UInt, j: UInt}, y: {i: UInt, j: UInt}} //7
wire b: {x: {i: UInt, j: UInt}, y: {i: UInt, j: UInt}} //7
a.x.i <= b.x.i //7+3+1 + 7 + 3 + 1 = 22
a.x.j <= b.x.j //22
a.y.i <= b.y.i //22
a.y.j <= b.y.j //22
```
totaling 102 Strings. After Lower Types (Low Firrtl):
```
wire a_x_i: UInt //1
wire a_x_j: UInt //1
wire a_y_i: UInt //1
wire a_y_j: UInt //1
wire b_x_i: UInt //1
wire b_x_j: UInt //1
wire b_y_i: UInt //1
wire b_y_j: UInt //1
a_x_i <= b_x_i //2
a_x_j <= b_x_j //2
a_y_i <= b_y_i //2
a_y_j <= b_y_j //2
```
Totaling 16 strings.

When compiling Boom, width inference is run on High Firrtl, Middle Firrtl, and Low Firrtl. It succeeds on High and Low, but fails in Middle. Fortunately, width inference is not necessary to run more than once, but we do it anyways just sort of because. I've made a change to add-chirrtl-check which removes the second running of width inference, and now Boom succeeds in compiling through Firrtl.

jackkoenig, 2016-07-26:
Nice work getting BOOM through!

I believe your analysis is on the right track but I'm not sure that this particular example is correct. I think we might need some profiling.

In your example, unless we (in the Firrtl compiler) are accidentally copying the Type of wires a and b, every later reference to those types should be exactly that, a reference, not a complete copy.

For example, in pseudo-Scala if I take some DefWire and create a reference to it:
```scala
val wire = DefWire(_, "myWire", BundleType(Seq(UIntType(32), SIntType(32)))
val ref = Reference(wire.name, wire.tpe)
```
I believe that the name and type of ref are not new objects, there shouldn't be any duplication. 

I could be wrong though or we could be accidentally creating lots of copies in the compiler. Any Scala gurus want to weigh in?

azidar, 2016-07-26:
Hmm yeah great point, you would expect all types to be references, not new objects. However, this isn't the case because, during Infer Widths, we walk the graph to find every width of every type of every expression (the mapr function). During this walk, we always return a new type based on applying our Width=>Width function. In essence, we are creating a massive number types because each reference to an existing type got replaced with a new type. This error only occurs during Boom's Middle Firrtl because only then are there enough types floating around (as the example demonstrates) to cause an issue.

A better way to structure this is to only replace the widths in the types of the wires/registers/ports etc but not the expressions. Then, do another infer types pass to propagate the type reference (instead of creating a bunch of new objects)

jackkoenig, 2016-07-26:
That makes sense, and this might explain why InferWidths takes substantially longer than most of the other passes. I think that solution is a great idea and would vastly speed up and reduce memory footprint of the compiler. 

To solve this issue in the general case (because I highly doubt this is the only place we create lots of duplicates), we could make Firrtl IR nodes automatically intern. I've done some poking around and found this: https://gist.github.com/vlfig/3757213. There might be other ways to do it as well. 

We could also create a performance regression that counts duplication of IR nodes and helps us identify and fix sources of it.

ccelio, 2016-07-29:
I think we can close this now. You can re-open (or make a new issue) if you'd like to catalog a desire to improve FIRRTL performance on this front.

closed 2016-07-29 by ccelio