chipsalliance/firrtl #202: What is the replacement for --noInlineMem + vlsi_mem_gen?

ben-k, 2016-07-25:
I'm hoping someone can clarify what the new strategy is for mapping a `SeqMem` to an SRAM in the Chisel3 + FIRRTL flow.

Chisel2 would elide the memory modules from the generated verilog and make a .conf helper file with the needed configurations.  This could then be used by a post-processing script to append the vendor-specific SRAM module definitions.

There doesn't appear to be an equivalent option in FIRRTL right now, so how is this supposed to be done?  Is the plan to re-implement --noInlineMem, or is there some more "Chisel-y" scheme?  If this already exists, please point me to it and I'll close this.

aswaterman, 2016-07-25:
This doesn't exist yet, but @yunsup is working on continuing the effort @jackkoenig started to create the FIRRTL equivalent of --noInlineMem.

ben-k, 2016-07-25:
That's good to hear - @jackkoenig, do you have an ETA for this?  It's fairly urgent for the tapeout work.

ben-k, 2016-08-03:
An update based on today's meeting: it's been decreed that @yunsup is implementing this (or is in charge of getting it implemented, anyway) and that it should be done in a week or two.

ben-k, 2016-09-07:
Per #265, I'm closing this.

closed 2016-09-07 by ben-k