chipsalliance/firrtl #209: FIRRTL emits inputs to printf(%c) that are more than 8 bits in size.

ccelio, 2016-07-30:
FIRRTL is emitting inputs to printf(%c) that are larger than 8-bits in size (due to programmer error).

Verilator complains about this, but other tool flows don't. 

> $fwriteof %c format of > 8 bit value

We should be able to catch this in FIRRTL and error out instead of emitting malformed Verilog.

jackkoenig, 2016-08-08:
Yeah this should be checked in "CheckWidths"

seldridge, 2019-01-22:
I thought this would be a 5-second fix...

There's some annoyance here in that when you find an `ir.Print` you need to determine the width of the expression associated with a specific delimiter. This has complications if we want to do it generally:

1. The expression refers to something defined earlier (`Utils.getWidth` only currently works on ground types)
2. The expression refers to something that needs to have its width inferred (this can't be placed in `CheckWidths` as it runs too early)
3. I'm lazy and know that `IRLookup` is in the PR queue... (A global `getWidth` is trivial with `IRLookup.tpe` yes @aziadr?)

So two approaches then:

1. Fix this for the immediate common case of ground types
2. Fix this for the general case of an expression

Also, the FIRRTL spec should then be updated to state that FIRRTL supports `%c`.

Thoughts?

seldridge, 2019-01-22:
Also, concrete test case:

```scala
class CharPrintfTester extends BasicTester {
  val x = 257.U
  printf("x=%c", x)
  stop()
}
class PrintfSpec extends ChiselFlatSpec {
  "A printf with a Char" should "run" in {
    assertTesterPasses { new CharPrintfTester }
  }
}
```

Chirrtl

```
;buildInfoPackage: chisel3, version: 3.2-SNAPSHOT, scalaVersion: 2.12.6, sbtVersion: 1.2.7
circuit CharPrintfTester : 
  module CharPrintfTester : 
    input clock : Clock
    input reset : UInt<1>
    output io : {}
    
    node _T = bits(reset, 0, 0) @[Printf.scala 35:9]
    node _T_1 = eq(_T, UInt<1>("h00")) @[Printf.scala 35:9]
    when _T_1 : @[Printf.scala 35:9]
      printf(clock, UInt<1>(1), "x=%c", UInt<9>("h0101")) @[Printf.scala 35:9]
      skip @[Printf.scala 35:9]
    node _T_2 = bits(reset, 0, 0) @[Printf.scala 36:7]
    node _T_3 = eq(_T_2, UInt<1>("h00")) @[Printf.scala 36:7]
    when _T_3 : @[Printf.scala 36:7]
      stop(clock, UInt<1>(1), 0) @[Printf.scala 36:7]
      skip @[Printf.scala 36:7]
    
```

Verilog:

```verilog
module CharPrintfTester(
  input   clock,
  input   reset
);
  wire  _T_1; // @[Printf.scala 35:9]
  assign _T_1 = reset == 1'h0; // @[Printf.scala 35:9]
  always @(posedge clock) begin
    `ifndef SYNTHESIS
    `ifdef PRINTF_COND
      if (`PRINTF_COND) begin
    `endif
        if (_T_1) begin
          $fwrite(32'h80000002,"x=%c",9'h101); // @[Printf.scala 35:9]
        end
    `ifdef PRINTF_COND
      end
    `endif
    `endif // SYNTHESIS
    `ifndef SYNTHESIS
    `ifdef STOP_COND
      if (`STOP_COND) begin
    `endif
        if (_T_1) begin
          $finish; // @[Printf.scala 36:7]
        end
    `ifdef STOP_COND
      end
    `endif
    `endif // SYNTHESIS
  end
endmodule
```