chipsalliance/firrtl #210: Additional Nested SubAccess Failure

seldridge, 2016-08-01:
Related to #105, I'm seeing a similar failure when doing multiple levels of dereferencing. The `WSubAccess` doesn't get properly handled and FIRRTL fails trying to handle it during the Lower Types pass.

The following FIRRTL produces the issue:
```txt
circuit AssignViaDeref : 
  module AssignViaDeref : 
    input clk : Clock
    input reset : UInt<1>
    output io : {a : UInt<8>, sel : UInt<1>}
    
    io is invalid
    reg table : {a : UInt<8>}[2], clk
    reg otherTable : {a : UInt<8>}[2], clk
    otherTable[table[UInt<1>("h01")].a].a <= UInt<1>("h00") @[Issues.scala 59:33] 
```

This was generated using the following Scala.
```scala
class MyEntry extends Bundle {
  val a = UInt(width = 8)
}

class MyEntryIO(n: Int) extends MyEntry {
  val sel = UInt(log2Up(n))
}

class AssignViaDeref extends Module {
  val io         = new MyEntryIO(2)
  val table      = Reg(Vec(2, new MyEntry))
  val otherTable = Reg(Vec(2, new MyEntry))

  otherTable(table(io.sel).a).a := UInt(0)
}
```

The truncated error is:
```
[INFO] Starting Lower Types
[error] (run-main-1) scala.MatchError: WSubAccess(WRef(table,VectorType(BundleType(ArrayBuffer(Field(a,Default,UIntType(IntWidth(8))))),2),RegKind(),MALE),UIntLiteral(1,IntWidth(1)),BundleType(ArrayBuffer(Field(a,Default,UIntType(IntWidth(8))))),MALE) (of class firrtl.WSubAccess)
scala.MatchError: WSubAccess(WRef(table,VectorType(BundleType(ArrayBuffer(Field(a,Default,UIntType(IntWidth(8))))),2),RegKind(),MALE),UIntLiteral(1,IntWidth(1)),BundleType(ArrayBuffer(Field(a,Default,UIntType(IntWidth(8))))),MALE) (of class firrtl.WSubAccess)
	at firrtl.passes.LowerTypes$.loweredName(LowerTypes.scala:62)
	at firrtl.passes.LowerTypes$.loweredName(LowerTypes.scala:64)
	at firrtl.passes.LowerTypes$.firrtl$passes$LowerTypes$$lowerTypesExp$1(LowerTypes.scala:149)
	at firrtl.passes.LowerTypes$$anonfun$firrtl$passes$LowerTypes$$lowerTypesExp$1$5.apply(LowerTypes.scala:154)
	at firrtl.passes.LowerTypes$$anonfun$firrtl$passes$LowerTypes$$lowerTypesExp$1$5.apply(LowerTypes.scala:154)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
...
```

azidar, 2016-08-01:
I've submitted a pull request with a fix, could you test it out on your code to make sure it works?

seldridge, 2016-08-01:
Just tested using #211 and I'm getting past this and all the way to Verilog generation using the behemoth source FIRRTL that motivated the MWE above. I don't have test cases in place for that Verilog, however... So, this is as good of a test as I can give it at this time.

Thanks for this (and for being so fast with it!).

closed 2016-08-02 by azidar