chipsalliance/firrtl #219: Support for asynchronously-resettable registers

ben-k, 2016-08-04:
FIRRTL is currently only capable of emitting registers with synchronous reset.  This is often OK, but sometimes an asynchronous reset is desirable (e.g., see our recent discussion at ucb-bar/rocket-chip#190, as well as [previous discussion](https://groups.google.com/forum/#!topic/chisel-users/4cc4SyB5mk8) from a variety of users).  Ultimately, the designer should have the flexibility to choose which type of reset to instantiate.  (See [this classic discussion](http://www.sunburst-design.com/papers/CummingsSNUG2002SJ_Resets.pdf) of the issue for more detail.)

Asynchronously-resettable flip-flops are a common verilog design idiom and are well-supported by EDA tools.  ucb-bar/chisel#549 has a good writeup of the desired feature; I've copied and cleaned up the code example here for convenience.  The option to select the reset polarity would also be nice.

```verilog
always @(posedge clk or posedge reset) begin
  if (reset) begin //async-resettable flops
    flop1 <= FLOP1_RESET_VALUE;
    flop2 <= FLOP2_RESET_VALUE;
  end else // D -> Q state updates for async-resettable flops
    flop1 <= flop1_next;
    flop2 <= flop2_next;
  end
end

always @(posedge clk) begin
  if (reset) begin //sync-resettable flops
    flopA <= FLOPA_RESET_VALUE;
    flopB <= FLOPB_RESET_VALUE;
  end else begin // D -> Q state updates for sync-resettable flops and flops with no reset
    flopA <= flopA_next;
    flopB <= flopB_next;
    flopC <= flopC_next; // flopC has no sync or async reset
  end
end
```

ben-k, 2016-08-04:
@azidar, per our discussion earlier.

aswaterman, 2016-08-04:
Async resets are just clocks, so if this feature is implemented, it should be as a generalization of reg that allows multiple clocks.

azidar, 2016-08-05:
To implement this feature, it seems one would only need to modify the Verilog emitter to pattern match for a Clock-type reset and emit the corresponding Verilog.

janboeye, 2018-03-06:
@azidar 
Why this feature is low priority? If we want to use Chisel in ASIC chip, this is a must-have feature. In this stage, is there any workaround for this?
Thanks a lot!

aswaterman, 2018-03-06:
As a workaround you can use blackboxes.  See for example: https://github.com/freechipsproject/rocket-chip/search?utf8=%E2%9C%93&q=asyncresetreg&type=

azidar, 2018-11-27:
The low priority tag was determined by user interest - if there is more significant interest, then we can certainly make it a high priority.

commit a33f3033 (2019-02-13): Add support for asynchronous reset registers

Fixes #219

* Adds AsyncResetType (similar to ClockType)
* Registers with reset signal of type AsyncResetType are async reset
  registers
* Registers with async reset can only be reset to literal values
PR #1011 (2019-02-14): Asynchronous Reset
I think this is more-or-less good to review!

Summary of changes:
* Adds AsyncResetType (similar to ClockType)
* Registers who's reset signal is of type AsyncResetType are async reset registers
* Registers with async reset can only be reset to literal values

I've probably missed a weird interaction somewhere but this does have several tests including an execution driven one.

@edwardcwang, per discussion with @aswaterman (and please pipe up if I miss[characterize] something), but it seems that the restriction of only allowing literals for async reset registers reset values  [... cut]
commit 2272044c (2019-02-14): Asynchronous Reset (#1011)

Fixes #219

* Adds AsyncResetType (similar to ClockType)
* Registers with reset signal of type AsyncResetType are async reset
  registers
* Registers with async reset can only be reset to literal values
* Add initialization logic for async reset registers
commit e3a15524 (2019-03-26): Asynchronous Reset (#1011)

Fixes #219

* Adds AsyncResetType (similar to ClockType)
* Registers with reset signal of type AsyncResetType are async reset
  registers
* Registers with async reset can only be reset to literal values
* Add initialization logic for async reset registers
closed 2019-02-14 by jackkoenig