chipsalliance/firrtl #221: Statement ordering (Request?)

shunshou, 2016-08-08:
I have been messing around with the following:

Create a new Module that instantiates an ExtModule + properly connect up ports of the Module to the ExtModule.

For the longest time, I couldn't figure out why I was getting this error:

Exception in thread "main" java.util.NoSuchElementException: key not found: mem_ext
	at scala.collection.MapLike$class.default(MapLike.scala:228)
	at scala.collection.AbstractMap.default(Map.scala:59)
	at scala.collection.MapLike$class.apply(MapLike.scala:141)
	at scala.collection.AbstractMap.apply(Map.scala:59)
	at firrtl.passes.InferTypes$.firrtl$passes$InferTypes$$infer_types_e$1(Passes.scala:173)

...

After printing out the body of a small FIRRTL module with a blackbox, I realized that this was due to the fact that I added a WDefInstance statement referring to the blackbox AFTER the Connect statements... 

(i.e. so the module body would have some List of Connect, Connect, Connect, WDefInstance instead of List of WDefInstance, Connect, Connect, Connect)

I realize that doing so is maybe not the most logical, but I guess I didn't really think list order would matter. Is there any way to make some kind of more descriptive error message? Or do the things in Passes.scala in such a way that order doesn't matter?

---

I'm not sure if the intention of FIRRTL passes is really for outsiders to be able to easily write them or if they're really only intended to be written by FIRRTL devs. If it's the latter case, then no worries about the error message... but it'd be super nice to have for FIRRTL noobs. 

Thanks! :)

jackkoenig, 2016-08-08:
At the very least, passes should be as easy to write as possible for outsiders. I think it would be hard to give more descriptive error messages *everywhere* because lots of passes make certain assumptions for simplicity, but providing a more extensive set of "Check" passes that pass writers can use to ensure their pass is correct seems like a really good idea. We have a few already like CheckHighForm which would have told you (I think), that you have a connection without a declaration. This is better but still not as descriptive as it could be. I think adding this (and other) more descriptive error messages to the Check passes and then encouraging pass writers to run all of the checks after their pass (at least in testing, which I actually think all passes should do).

shunshou, 2016-08-08:
One more thing -- just kind of general advice to other people starting out, but it's much easier to figure out what's going on (and what to add to passes) by writing your own FIRRTL module (what you expect your pass to transform the code into) and having it print out the module body. 

It's way easier to see how PrimOps, etc. needed to be added to the statement list that way :p

shunshou, 2016-08-08:
Oh and regarding CheckHighForm or similar, is there any way to get into a little more detail? I do add the instance in the List of Statements, it just happens to have happened AFTER the connections... so a simple connection without a declaration error is kind of weird, because it exists?

azidar, 2016-08-08:
Gender is used to judge the absolute directions of expressions (source or sink). Some passes require knowing this knowledge, others don't. I imagine they do not matter for your case. If you don't know what gender something should have, use the UnknownGender object and run ResolveGenders after your pass. This will correctly replace all UnknownGender's (and incorrectly labeled genders, I believe) with the correct Gender.

A Connect's lhs should always be Female, and the rhs should always be Male. Subexpressions of the rhs/lhs can be either gender, however (depends on the type of the expressions)

shunshou, 2016-08-08:
Thanks for the clarification. :) I probably also should've stared at the FIRRTL spec more thoroughly.

azidar, 2016-08-09:
No problem, happy to help :)