chipsalliance/firrtl #224: Generalization of SMem

shunshou, 2016-08-08:
So, from previous tapeout experience with Chisel SeqMems, it's pretty apparent that certain flavors of memory just won't work unless you thread a reset/init signal to them. If that's the case, shouldn't reset be part of the memory spec? Additionally, foundry memory macros typically have a bunch of other setup ports. There might be certain cases where these settings should be runtime controllable, but since the memory spec is rigid, it's not easy to accommodate these customizations. Have people thought about expanding the SMem specs with maybe some optionable IO that allows for, say, customizable reset + settings? 

Also, was there a particular reason why why Mems weren't just directly blackboxed?

aswaterman, 2016-08-08:
The intent was to have backend-specific FIRRTL passes to handle this kind of thing.  Even the reset/init pins are arguably too backend-specific to expose in a generic way (they don't always exist, or vary in how/whether they're synchronized to the clock).

The desire to isolate this detail from the RTL led to not black-boxing memories directly, as the Chisel code would need to know about the backend-specific ports and so wouldn't be generic.

azidar, 2016-08-13:
Is it ok to close this issue?

shunshou, 2016-08-13:
Yep

closed 2016-08-13 by shunshou