chipsalliance/firrtl #230: RemoveAccesses exception when indexing 0-element vec

aswaterman, 2016-08-12:
The following circuit causes RemoveAccesses to fail:

    circuit bar :
      module bar :
        output foo : UInt<1>[0]
        foo[UInt(0)] <= UInt(0)

I think this should be legal and the behavior should match the non-power-of-2 out-of-bounds case, e.g. indexing a 5-element array with dynamic index 7.

    Exception in thread "main" java.lang.IndexOutOfBoundsException: 0
	at scala.collection.mutable.ResizableArray$class.apply(ResizableArray.scala:43)
	at scala.collection.mutable.ArrayBuffer.apply(ArrayBuffer.scala:48)
	at firrtl.passes.RemoveAccesses$.removeFemale$1(RemoveAccesses.scala:114)
	at firrtl.passes.RemoveAccesses$.firrtl$passes$RemoveAccesses$$onStmt$1(RemoveAccesses.scala:145)
	at firrtl.passes.RemoveAccesses$$anonfun$4.apply(RemoveAccesses.scala:146)
	at firrtl.passes.RemoveAccesses$$anonfun$4.apply(RemoveAccesses.scala:146)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.Mappers$StmtMagnet$$anon$4.map(Mappers.scala:44)
	at firrtl.Mappers$StmtMap.map(Mappers.scala:95)
	at firrtl.passes.RemoveAccesses$.firrtl$passes$RemoveAccesses$$onStmt$1(RemoveAccesses.scala:146)
	at firrtl.passes.RemoveAccesses$$anonfun$4.apply(RemoveAccesses.scala:146)
	at firrtl.passes.RemoveAccesses$$anonfun$4.apply(RemoveAccesses.scala:146)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.Mappers$StmtMagnet$$anon$4.map(Mappers.scala:44)
	at firrtl.Mappers$StmtMap.map(Mappers.scala:95)
	at firrtl.passes.RemoveAccesses$.firrtl$passes$RemoveAccesses$$onStmt$1(RemoveAccesses.scala:146)
	at firrtl.passes.RemoveAccesses$.firrtl$passes$RemoveAccesses$$remove_m$1(RemoveAccesses.scala:151)
	at firrtl.passes.RemoveAccesses$$anonfun$6.apply(RemoveAccesses.scala:156)
	at firrtl.passes.RemoveAccesses$$anonfun$6.apply(RemoveAccesses.scala:154)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.passes.RemoveAccesses$.run(RemoveAccesses.scala:154)
	at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:48)
	at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:48)
	at firrtl.Utils$.time(Utils.scala:57)
	at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:48)
	at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:46)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.SimpleRun$class.run(LoweringCompilers.scala:45)
	at firrtl.HighFirrtlToMiddleFirrtl.run(LoweringCompilers.scala:104)
	at firrtl.HighFirrtlToMiddleFirrtl.execute(LoweringCompilers.scala:117)
	at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:71)
	at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:70)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.Compiler$class.compile(Compiler.scala:70)
	at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:198)
	at firrtl.Driver$.compile(Driver.scala:60)
	at firrtl.Driver$.run(Driver.scala:173)
	at firrtl.Driver$.main(Driver.scala:96)
	at firrtl.Driver.main(Driver.scala)

azidar, 2016-08-15:
What would the expanded form of ```output foo: UInt[0]``` be?

aswaterman, 2016-08-15:
No output at all would be consistent with the current flattening scheme for vectors.  I believe that part is already correctly implemented.

ekiwi, 2019-10-28:
Just ran the example from Andrew with the latest `master` branch of `firrtl`:

```
Exception in thread "main" firrtl.passes.CheckTypes$IndexTooLarge: : [module bar]  Index with value 0 is too large.
```

This seems to make sense since the Vec has zero elements and thus there is no element zero.

azidar, 2019-10-28:
This example is still broken:

```
circuit bar :
  module bar :
    output foo : UInt<1>[0]
    input x: UInt<1>
    foo[x] <= UInt(0)
```

with stack trace:

```
Exception in thread "main" firrtl.FirrtlInternalException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
	at firrtl.Utils$.error(Utils.scala:431)
	at firrtl.Utils$.throwInternalError(Utils.scala:166)
	at firrtl.stage.FirrtlStage.run(FirrtlStage.scala:35)
	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:32)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
	at firrtl.options.Translator.transform(Phase.scala:140)
	at firrtl.options.Translator.transform$(Phase.scala:140)
	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
	at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:46)
	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
	at scala.collection.immutable.List.foldLeft(List.scala:89)
	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:46)
	at logger.Logger$.$anonfun$makeScope$2(Logger.scala:168)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
	at logger.Logger$.makeScope(Logger.scala:166)
	at firrtl.options.Stage.transform(Stage.scala:46)
	at firrtl.options.Stage.execute(Stage.scala:57)
	at firrtl.options.StageMain.main(Stage.scala:70)
	at firrtl.stage.FirrtlMain.main(FirrtlStage.scala)
Caused by: java.lang.UnsupportedOperationException: empty.head
	at scala.collection.immutable.Vector.head(Vector.scala:189)
	at firrtl.passes.RemoveAccesses.removeFemale$1(RemoveAccesses.scala:121)
	at firrtl.passes.RemoveAccesses.onStmt$1(RemoveAccesses.scala:154)
	at firrtl.passes.RemoveAccesses.$anonfun$run$10(RemoveAccesses.scala:155)
	at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:238)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at scala.collection.TraversableLike.map(TraversableLike.scala:238)
	at scala.collection.TraversableLike.map$(TraversableLike.scala:231)
	at scala.collection.immutable.List.map(List.scala:298)
	at firrtl.ir.Block.mapStmt(IR.scala:397)
```

lijinpei, 2019-12-15:
May I ask what's the expected expanded form of this example:
```fir
circuit bar :
  module bar :
    output foo : UInt<1>[0]
    input x: UInt<1>
    foo[x] <= UInt(0)
```
A. Compile time error message(but not compiler throw exception)
B. effectively this(because vector index automatically wrap arounds and zero-length dead vectors are eliminated):
```fir
circuit bar :
  module bar :
    input x: UInt<1>
```

commit c47e6d31 (2020-03-13): Fix dynamic SubAccess of zero-length vectors

* Add new ZeroLengthVecs pass that occurs before RemoveAccesses
* Include this in stage.Forms.MidForm
* Add to LoweringCompilersSpec as an Add(...) patch for High -> Mid
* Fixes #230
commit c77e5f04 (2020-03-14): Fix dynamic SubAccess of zero-length vectors

* Fixes #230
* Add new ZeroLengthVecs pass that occurs before RemoveAccesses
* Include this in stage.Forms.MidForm
* Add to High->Mid order in compiler test based on @seldridge feedback
commit 1a03e635 (2020-04-07): Fix dynamic SubAccess of zero-length vectors (#1450)

* Fix dynamic SubAccess of zero-length vectors

* Fixes #230
* Add new ZeroLengthVecs pass that occurs before RemoveAccesses
* Include this in stage.Forms.MidForm
* Add to High->Mid order in compiler test based on @seldridge feedback

* Use validif to produce out-of-bounds value in ZeroLengthVecs

* Update scaladoc

* Fix test imports
commit e40640e5 (2020-04-08): Fix dynamic SubAccess of zero-length vectors (#1450)

* Fix dynamic SubAccess of zero-length vectors
* Fixes #230
* Add new ZeroLengthVecs pass that occurs before RemoveAccesses
* Include this in stage.Forms.MidForm
* Add to High->Mid order in compiler test based on @seldridge feedback
* Use validif to produce out-of-bounds value in ZeroLengthVecs
* Update scaladoc
* Fix test imports
* Delete DependencyAPI stuff for backport

(cherry picked from commit 1a03e6356e451136d522d5a9acba374dd8972b24)
closed 2020-04-07 by mergify[bot]