chipsalliance/firrtl #232: connect lhs should be an id, not an exp

nturley, 2016-08-15:
Not a big deal, but I've been messing with the grammar and I noticed that the [left hand side of a connection statement](https://github.com/ucb-bar/firrtl/blob/cca37c46fc0848f5dbf5f95ba60755ed6d60712b/src/main/antlr4/FIRRTL.g4#L115) is an exp. I think it should be an id, instead. No other type of exp makes any sense.

I think that's true for the partial connect and the invalid statement as well. Making a constant invalid wouldn't make any sense.

azidar, 2016-08-15:
Thanks for checking out Firrtl! However, the left side of a connection statement can be more than an id, namely it can also be a sub-index ("a[0]"), a sub-access ("a[b]"), or a sub-field ("a.c"). As you imply, however, there are expressions which are not legal to have on the left hand side of an expression (mux, constant, primitive operation, validif). These invariants are checked, just not by the parser.

nturley, 2016-08-16:
ah. That makes more sense. I've only been playing with lofirrtl so I forgot about those.

closed 2016-08-16 by nturley