chipsalliance/firrtl #234: Better Usage message(s) for cmd line args

mwachs5, 2016-08-15:
Can we add to the FIRRTL README some indication of how to figure out what the FIRRTL cmd line arguments are. 

I believe the correct action is:

$FIRRTL --help

However, this doesn't print out all the arguments.  For example the --inline flag has no explanation, and the new flags introduced on @shunshou's sub_mem_bb don't show up if you use --help. 

Isn't there some way (similar to python's argparse) that can handle the argument parsing and documentation in one pass?

Giving an unknown argument should also print the correct usage in addition to complaining about the unsupported argument.

kammoh, 2016-08-15:
I can chip in scopt (https://github.com/scopt/scopt) for arg parsing.

azidar, 2016-08-15:
That would be awesome @kammoh! For now I'll push a fix that updates the usage string and prints it on incorrect args, and then feel free to submit the PR and I'll merge it.

kammoh, 2016-08-15:
I could probably hack up a PR in a few days if that's cool and the added
dependency is fine.

azidar, 2016-08-15:
Yeah, the added dependency its fine. Thanks!

closed 2016-11-16 by azidar