chipsalliance/firrtl #250: Requirement failed in constant propagation

zhemao, 2016-08-24:
I'm got the following FIRRTL error when trying to do something in RocketChip. When I look at the require on line 236 in ConstProp.scala, it's not very clear what it means. Is that requirement ever meant to be triggered?

    Exception in thread "main" java.lang.IllegalArgumentException: requirement failed
	at scala.Predef$.require(Predef.scala:207)
	at firrtl.passes.ConstProp$.constPropPrim(ConstProp.scala:236)
	at firrtl.passes.ConstProp$.firrtl$passes$ConstProp$$constPropExpression$1(ConstProp.scala:274)
	at firrtl.passes.ConstProp$$anonfun$1.apply(ConstProp.scala:272)
	at firrtl.passes.ConstProp$$anonfun$1.apply(ConstProp.scala:272)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.immutable.List.map(List.scala:285)
	at firrtl.Mappers$ExpMagnet$$anon$5.map(Mappers.scala:111)
	at firrtl.Mappers$ExpMap.map(Mappers.scala:144)
	at firrtl.passes.ConstProp$.firrtl$passes$ConstProp$$constPropExpression$1(ConstProp.scala:272)
	at firrtl.passes.ConstProp$$anonfun$firrtl$passes$ConstProp$$constPropStmt$1$3.apply(ConstProp.scala:289)
	at firrtl.passes.ConstProp$$anonfun$firrtl$passes$ConstProp$$constPropStmt$1$3.apply(ConstProp.scala:289)
	at firrtl.Mappers$StmtMagnet$$anon$3.map(Mappers.scala:53)
	at firrtl.Mappers$StmtMap.map(Mappers.scala:95)
	at firrtl.passes.ConstProp$.firrtl$passes$ConstProp$$constPropStmt$1(ConstProp.scala:289)
	at firrtl.passes.ConstProp$$anonfun$firrtl$passes$ConstProp$$constPropStmt$1$1.apply(ConstProp.scala:289)
	at firrtl.passes.ConstProp$$anonfun$firrtl$passes$ConstProp$$constPropStmt$1$1.apply(ConstProp.scala:289)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.immutable.List.map(List.scala:285)
	at firrtl.Mappers$StmtMagnet$$anon$4.map(Mappers.scala:44)
	at firrtl.Mappers$StmtMap.map(Mappers.scala:95)
	at firrtl.passes.ConstProp$.firrtl$passes$ConstProp$$constPropStmt$1(ConstProp.scala:289)
	at firrtl.passes.ConstProp$$anonfun$firrtl$passes$ConstProp$$constPropStmt$1$1.apply(ConstProp.scala:289)
	at firrtl.passes.ConstProp$$anonfun$firrtl$passes$ConstProp$$constPropStmt$1$1.apply(ConstProp.scala:289)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.Mappers$StmtMagnet$$anon$4.map(Mappers.scala:44)
	at firrtl.Mappers$StmtMap.map(Mappers.scala:95)
	at firrtl.passes.ConstProp$.firrtl$passes$ConstProp$$constPropStmt$1(ConstProp.scala:289)
	at firrtl.passes.ConstProp$$anonfun$firrtl$passes$ConstProp$$constPropStmt$1$1.apply(ConstProp.scala:289)
	at firrtl.passes.ConstProp$$anonfun$firrtl$passes$ConstProp$$constPropStmt$1$1.apply(ConstProp.scala:289)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.immutable.List.map(List.scala:285)
	at firrtl.Mappers$StmtMagnet$$anon$4.map(Mappers.scala:44)
	at firrtl.Mappers$StmtMap.map(Mappers.scala:95)
	at firrtl.passes.ConstProp$.firrtl$passes$ConstProp$$constPropStmt$1(ConstProp.scala:289)
	at firrtl.passes.ConstProp$.firrtl$passes$ConstProp$$constPropModule(ConstProp.scala:292)
	at firrtl.passes.ConstProp$$anonfun$3.apply(ConstProp.scala:299)
	at firrtl.passes.ConstProp$$anonfun$3.apply(ConstProp.scala:297)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.passes.ConstProp$.run(ConstProp.scala:297)
	at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:48)
	at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:48)
	at firrtl.Utils$.time(Utils.scala:57)
	at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:48)
	at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:46)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.SimpleRun$class.run(LoweringCompilers.scala:45)
	at firrtl.EmitVerilogFromLowFirrtl.run(LoweringCompilers.scala:142)
	at firrtl.EmitVerilogFromLowFirrtl.execute(LoweringCompilers.scala:154)
	at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:71)
	at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:70)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.Compiler$class.compile(Compiler.scala:70)
	at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:198)
	at firrtl.Driver$.compile(Driver.scala:60)
	at firrtl.Driver$.run(Driver.scala:173)
	at firrtl.Driver$.main(Driver.scala:96)
	at firrtl.Driver.main(Driver.scala)

zhemao, 2016-08-24:
Oh, I see now, this is making sure that when you extract bits from a larger word that the MSB is larger than the LSB. Shouldn't this print a better error message?

shunshou, 2016-08-24:
In general, it'd be great if errors could report the node name that failed (say, for expression matching). I've been having to go in and modify firrtl code with printlns to debug firrtl errors...

aswaterman, 2016-08-24:
In this case, this really is an internal compiler error: a checking pass
should validate that Bits has valid arguments.  ConstProp probably
shouldn't even have that require().

But I completely agree with Angie; when an internal compiler error does
occur, some basic diagnostics would be helpful (to the compiler writers,
not so much to the users).

azidar, 2017-12-15:
Fixed.

closed 2017-12-15 by azidar