chipsalliance/firrtl #251: Memory read-under-write

shunshou, 2016-08-25:
Two parts to my question:

1) Should read-under-write really be built into DefMemory vs. a separate annotation? It probably makes sense to keep it as part of DefMemory (since, theoretically, it *should* be defined), although it really is an annotation. 

2) There's no Chisel -> Firrtl support for it, correct? I assume that should go along with the Chisel3 issue #264 regarding the need for annotation API.

The particular use-case:

As I've stated before in some Chisel2 related issue, BRAMs are generally only properly inferred on FPGAs if you register the data output (as opposed to the read address). An "FPGA memory" Firrtl pass needs to relocate the address register to the data output register (even though the FPGA IP core internally moves the register back), but the read-under-write behavior needs to be consistent when such a pass does this.

azidar, 2016-08-25:
Re 1): Yes it must built into DefMemory because if we are writing an interpreter defining the behavior of DefMemory, it is incomplete without defining this behavior.

Re 2): Not to my knowledge.

schoeberl, 2016-08-29:
> Two parts to my question:
>
> 1) Should read-under-write really be built into DefMemory vs. a separate annotation? It probably makes sense to keep it as part of DefMemory (since, theoretically, it [...]

My experience is the other way round: the address inputs (read and write) and the write data need to be in registers. Data out may be combinational.

In FPGA memories often read during write is undefined. A synthesize tool might add forwarding from a memory definition. At least I have seen this of VHDL described memories and Quartus. However, I prefer to be on the save side and add the forwarding myself.

The endless memory topic ;-)

Cheers,
Martin

aswaterman, 2016-08-29:
One of the options (the default option, I'd say) should be that the read returns undefined data, to cater to ASIC RAMs, for which all of the well-defined options require extra circuitry and delay.

schoeberl, 2016-09-08:
I agree that this should be the default option. It is also quite common in FPGAs. However, I would not know how this can be specified in Verilog (or VHDL).

Cheers,
Martin

closed 2017-03-16 by azidar