chipsalliance/firrtl #254: FixedPoint operations have a few problems

chick, 2016-08-26:
It does not seem possible to express a ```Fixed``` with a negative binaryPoint
Multiplication should not shift either argument
Other problems to come

chick, 2016-08-29:
I just pushed a new test FixePointMathSpec that fails on a number of the problems.
There may be some problems in test with subtraction

shunshou, 2016-08-30:
Just want to make a note here -- for the purposes of range propagation, +%, -% range should expand to next power of 2 - 1. I didn't do that in my old ChiselDSP stuff.

chick, 2016-08-30:
I added a new Issue 255 for Range Propagation and put this comment there
too.

Chick Markley
Staff Programmer -- Aspire Lab
University of California, Berkeley
583-2 Soda Hall, Berkeley, CA 94720-1776
chick@berkeley.edu

closed 2016-09-13 by azidar