chipsalliance/firrtl #261: Optimizations fail to eliminate redundant last connect semantics

colinschmidt, 2016-09-02:
In the example below we see a simple piece of chisel code that generates correct but redundant code. A CAD tool has identified this as a warning:

**Statement unreachable (Branch condition impossible to meet).**

I'm not sure if this is DCE or CSE but I think it is reasonable to catch it.

**Chisel:**
```
when (issue) {
    op(vidx_tail) := opq.io.deq.bits
    op(vidx_tail).vlen := vlen_tail
```

**FIRRTL:**
```
    when issue : @[dcc-mem.scala 657:16] 
      op[map.io.use.bits.vidx] <- opq.io.deq.bits @[dcc-mem.scala 658:19] 
      op[map.io.use.bits.vidx].vlen <= vlen_tail @[dcc-mem.scala 659:24]
```

**LOW FIRRTL:**
```
    wire GEN_28 : UInt<12> @[dcc-mem.scala 658:19]
    node GEN_345 = mux(eq(UInt<1>("h0"), map.io_use_bits_vidx), GEN_28, op_0_vlen) @[dcc-mem.scala 658:19]
...
    node GEN_389 = mux(eq(UInt<1>("h0"), map.io_use_bits_vidx), GEN_50, GEN_345) @[dcc-mem.scala 659:24]
...
    node GEN_394 = mux(issue, GEN_389, op_0_vlen) @[dcc-mem.scala 657:16]
```

**VERILOG**:
```
      if(issue) begin
        if(1'h0 == map_io_use_bits_vidx) begin
          op_0_vlen <= GEN_50;
        end else begin
          if(1'h0 == map_io_use_bits_vidx) begin
            op_0_vlen <= GEN_28;
          end
        end
      end
```

aswaterman, 2016-09-02:
Handling this completely is a satisfiability problem.  We could pattern match certain cases (this one's easy) but should be disciplined about how we do so.

colinschmidt, 2016-09-02:
I'm not super worried about this but the tool is giving me hundreds of these warnings and it seems like this could very well be an RTL error for some cases, but clearly not ones of this variety.