chipsalliance/firrtl #262: Verilog has lint warnings in vcs again

colinschmidt, 2016-09-02:
FYI, recent versions of VCS emit a lint warning about [this](https://github.com/ucb-bar/firrtl/commit/4e69f7fe305eb67de0b50713d298869f64d889f3) if statement, because it contains no statements:

```
Lint-[NS] Null statement
/scratch/colins/rocket-chip/vsim/generated-src/TestHarness.ISCA2016Config.v, 740669
  Null statement is used in following verilog source.
```

For now I've just add +lint=noNS to my command line arguments but it would be cool if we got back to no vcs warnings

aswaterman, 2016-09-02:
Could you provide a gist of the surrounding Verilog?

aswaterman, 2016-09-02:
Nevermind, I saw it.  I can't figure out why it's problematic, though... can you propose how you'd change the Verilog output to make VCS happy?

colinschmidt, 2016-09-02:
Surprisingly simply removing the semicolon stops the lint warning for me :)

colinschmidt, 2016-09-02:
I can submit a PR later tonight.

aswaterman, 2016-09-02:
Careful here... the block may effectively be empty, based upon the settings of the various RANDOMIZE_ options.  So if you remove the semicolon, it can become a syntax error.

Ideally you'd just have an explicit null statement (kind of like "(void) 0;" in C, which compilers know to not warn about since it's explicitly doing nothing).

colinschmidt, 2016-09-04:
It turns out the recommended way to avoid this null statement problem is with an empty begin end block.
Applying this removes the warnings with and without randomize* flags.

Submitting a PR now. Thanks for the help.

PR #264 (2016-09-05): Change null statement to empty begin end
this eliminates warnings in recent versions of VCS

Fixes #262
closed 2016-09-05 by azidar