chipsalliance/firrtl #269: Better error messages for uninferred widths

aswaterman, 2016-09-07:
If you make a reg with no width, and fail to assign to it, CheckWidths throws an exception. Since this is not an internal compiler error, but the result of incorrect user input, we should print a more helpful message. (Sorry, I'm on my phone so can't provide a failing test case.)

aswaterman, 2016-09-07:
Alternatively, we could define this case to be legal, and infer widths of 1 (or 0, once we support that).

azidar, 2017-12-15:
This should be fixed with the new constraint solver.

seldridge, 2018-10-25:
Is this a test case like what you were thinking of @aswaterman?
```scala
  "Reg Foo.x" should "error with a good message" in {
    val passes = Seq(
      ToWorkingIR,
      ResolveKinds,
      InferTypes,
      CheckTypes,
      InferWidths,
      CheckWidths)
    val input =
      """|circuit Foo :
         |  module Foo :
         |    input clock: Clock
         |    reg x: UInt, clock
         |""".stripMargin
    executeTest(input, Nil, passes)
  }
```

This then gives you the following:
```
[info] Reg Foo.x
[info] - should error with a good message *** FAILED ***
[info]   firrtl.passes.CheckWidths$UninferredWidth: : [module Foo]  Uninferred width.
```

This seems sane to me as is, but it does point towards a FIRRTL problem and not a user error.

Note: the same problem exists for unconnected wires without a specified width.

Two complementary approaches that I see for making this better:
  1. All `Check*` exceptions provide a full `Named` (or instance annotation-style `Target`) that tells you exactly what thing was uninferred (`Uninferrable width for 'Foo/x'`)
  2. Add the text `"Did you forget to assign to it?"` to an uninferred width exception

The default width idea (e.g., a default constraint that all widths are greater than -1) would seem to cause havoc as then no widths are uninferrable, but many unspecified widths may be inferred incorrectly.

Does this seem reasonable?

edwardcwang, 2018-10-25:
Something like `Uninferred width for 'Foo/x'` seems useful to help the user avoid having to guess what was uninferred. (If FileInfo is present, isn't that shown in current error messages already?)

seldridge, 2018-10-25:
The `Info` is available to the exception, but it's not printed for whatever reason. That does assume that source locators were included and that's a very Chisel-centric thing. For the case of manually written FIRRTL (or at least FIRRTL passed in from the command line/not coming from a file), that won't be available.

PR #932 (2018-11-05): Better error message for UninferredWidth exception
This improves `CheckWidths` pass by passing along the `Target` that is currently being examined. This enables a better` UninferredWidth` exception message indicating what `Target` can't be inferred. This improves over the current exception message which only tells you what module couldn't be inferred.

Fixes #269 

Consider the following test case included in this PR:
```
circuit Foo :
  module Foo :
    input clock : Clock
    inst bar of Bar
  module Bar :
    wire a: { b : UInt<1>, c : { d : UInt<1>, e : UInt } }
```

`a.c.e` in `Bar` is un [... cut]
closed 2018-11-05 by seldridge