chipsalliance/firrtl #274: Proposed solution for wiring configuration bits to blackboxed SeqMems

ben-k, 2016-09-07:
**The Problem**

Vendor-supplied memories may have one or more configuration pins that cannot be statically set, but need to be toggled by some other part of the system.  The canonical example of this is the `init` pin that was emitted by the old `rocketChipBackend`, but there may be others as well.  Chisel3 and Firrtl do not currently have a system for blackboxed memories to include arbitrary IOs, and even if they did, there would be no way to wire to them.

**Proposed Solution**

This is a two-parter.  First, there ought to be some way of passing in a configuration option to the `replseqmem` pass(es) that allows the user to name extra ports in the generated blackboxes.

Second, these extra ports should connect by name to wires in the same module.  So, for instance, if my code snippet looked like this:

```scala
val init = io.sram_conf.init // Set init value elsewhere in the design
val my_mem = SeqMem(...)
```

Then if the configuration file for the `replseqmem` pass specified that all memory blackboxes should have an init pin, then init as defined in the above source would be wired into the emitted `my_mem` blackbox.

**Possible Issue**

Would the wires just be optimized away before Firrtl could match them by name?  I don't know enough about the Firrtl pass ordering to answer this.

I'm open to other suggestions about how to implement this, but the feature is tapeout-critical so I hope we can come to consensus soon.

shunshou, 2016-09-07:
Alternative is to bring all of the memory configuration ports up to the top module and inject/hook up a control block into it via FIRRTL. [Should not change rocket], but each tapeout might have its own custom control block that the pass would need to insert.

jackkoenig, 2016-09-07:
The custom control block could still be written in Chisel, just as separate from rocket-chip and wired up just via Firrtl pass

closed 2016-11-16 by azidar