chipsalliance/firrtl #286: Unconditional SeqMem reads turn into never executed reads

terpstra, 2016-09-08:
Currently, if there is no 'when' infront of the address heading into a SeqMem read address, firrtl creates a constant '0' for the read enable. It should be a constant '1'!

eg:
    val mem = SeqMem(1 << addrBits.size, Vec(beatBytes, Bits(width = 8)))
    val bar = mem.read(counter) // read address never follows counter!

terpstra, 2016-09-08:
@shunshou I've been told this was probably broken by the new firrtl SRAM extraction code.

shunshou, 2016-09-08:
Let me look into it. Can you send me an e-mail with Chisel generated FIRRTL/chirrtl for whatever config you're using? (Before its run through actual FIRRTL).

Is this for fpga configuration?

shunshou, 2016-09-08:
I'm also wondering if this is a problem with the Chirrtl to high FIRRTL pass, because I recall that it initializes memory ports to 0 and then adds in logic from whens. But if it hasn't been a problem before, then maybe it's my pass doing something wonky.

shunshou, 2016-09-08:
This is an issue that's outside of my transform, I think.

```
class GCD extends Module {
  val io = new Bundle {
    val wen = Bool(INPUT)
    val in = Bits(INPUT,1)
    val counter = UInt(INPUT, 2)
    val out = Bits(OUTPUT,1)
  }
  val mem = SeqMem(4, Bits(width = 1))
  val temp = io.counter + UInt(1)
  val bar = mem.read(temp) 
  when (io.wen) {
    mem.write(io.counter,io.in)
  }
  io.out := bar
}
```

Produces the CHIRRTL:

```
circuit GCD : 
  module GCD : 
    input clk : Clock
    input reset : UInt<1>
    output io : {flip wen : UInt<1>, flip in : UInt<1>, flip counter : UInt<2>, out : UInt<1>}
    
    io is invalid
    smem mem : UInt<1>[4] @[GCD.scala 14:19] 
    node T_7 = add(io.counter, UInt<1>("h01")) @[GCD.scala 15:25] 
    node temp = tail(T_7, 1) @[GCD.scala 15:25] 
    read mport bar = mem[temp], clk
    when io.wen : @[GCD.scala 18:17] 
      write mport T_8 = mem[io.counter], clk
      T_8 <= io.in
      skip @[GCD.scala 18:17] 
    io.out <= bar @[GCD.scala 21:10] 
```

Which produces High FIRRTL:

```
circuit GCD :
  module GCD :
    input clk : Clock
    input reset : UInt<1>
    output io : { flip wen : UInt<1>, flip in : UInt<1>, flip counter : UInt<2>, out : UInt<1>}
  
    io is invalid
    mem mem : @[GCD.scala 14:19]
      data-type => UInt<1>
      depth => 4
      read-latency => 1
      write-latency => 1
      reader => bar
      writer => T_8
      read-under-write => undefined
    mem.bar.addr is invalid @[GCD.scala 14:19]
    mem.bar.clk is invalid @[GCD.scala 14:19]
    mem.bar.en <= UInt<1>("h0") @[GCD.scala 14:19]
    mem.T_8.addr is invalid @[GCD.scala 14:19]
    mem.T_8.clk is invalid @[GCD.scala 14:19]
    mem.T_8.en <= UInt<1>("h0") @[GCD.scala 14:19]
    mem.T_8.data is invalid @[GCD.scala 14:19]
    mem.T_8.mask <= UInt<1>("h0") @[GCD.scala 14:19]
    node T_7 = add(io.counter, UInt<1>("h1")) @[GCD.scala 15:25]
    node temp = tail(T_7, 1) @[GCD.scala 15:25]
    mem.bar.addr <= temp
[... 89 lines cut ...]
    mem.bar.clk <= clk
    mem.bar.en <= UInt<1>("h1")
    when io.wen : @[GCD.scala 18:17]
      mem.T_8.addr <= io.counter
      mem.T_8.clk <= clk
      mem.T_8.en <= UInt<1>("h1")
      mem.T_8.data <= io.in
      mem.T_8.mask <= UInt<1>("h1")
      skip
    io.out <= mem.bar.data @[GCD.scala 21:10]
```

Note specifically, as before, earlier in the High FIRRTL, there is a line:
` mem.bar.en <= UInt<1>("h0") @[GCD.scala 14:19]`

But this time, later in the code, there is a line:
`mem.bar.en <= UInt<1>("h1")`

Because of a pass that resolves last connect semantics, the generated Verilog will have the read enable assigned to 1 in this case, which is right... 

@terpstra it'd be great to know if this was an issue for you before or if it's something that recently popped up. I can probably try to find a way around this, but I don't really like the current way memory ports are inferred from the CHIRRTL. 

Heck, I kind of wish people could think a little more about revamping the Chisel Emitter, because, at least for memories, it feels like you have more info when you're coding your Chisel project, and then it's getting dropped by the time Chisel emits CHIRRTL, and then that first FIRRTL pass has to overcomplicate the process of re-inferring this extra info (and possibly making mistakes, as in this case). 

Opinions? @azidar @donggyukim @jackkoenig and other people who might have an opinion?

shunshou, 2016-09-08:
Ok, from the looks of it, even though read en was 0 in one case or 1 in the other case, the Verilog didn't actually use it for anything when you just inferred registers... but if you try with actual SRAMs, the en's actually need to be right...

@donggyukim I'm a little confused. How does the `s.exps(0)` match part work/why is it necessary?

When the address is generated internally, it's *Reference(temp,UnknownType)*. When it's an IO, it's *SubField(Reference(io,UnknownType),counter,UnknownType)*.

This affects the match in a way that you probably didn't intend? When it's a subfield, it'll properly add en = 1, and that would explain why one version is ok and the other isn't. 

```
case MRead => {
                        repl(s.name) = DataRef(SubField(Reference(s.mem,ut),s.name,ut),"data","data","blah",false)
                        addrs += "addr"
                        clks += "clk"
                        s.exps(0) match {
                           case e: Reference if smems(s.mem) =>
                              raddrs(e.name) = SubField(SubField(Reference(s.mem,ut),s.name,ut),"en",ut)
                           case _ => ens += "en"
                        }
                     }
```

I can't really follow why `raddrs(e.name) = SubField(SubField(Reference(s.mem,ut),s.name,ut),"en",ut)` is necessary. Basically, if you get rid of the whole match part and replace with `ens += "en"`, you get the right result...

terpstra, 2016-09-08:
Wow. Thanks for the rapid response; sorry I didn't see your replies earlier.

On Wed, Sep 7, 2016 at 10:10 PM, Angie Wang <notifications@github.com>
wrote:

Which produces High FIRRTL:
>
>     mem.bar.en <= UInt<1>("h0") @[GCD.scala 14:19]
>
> Note that the original Verilog gets the read address from an internally generated signal [temp, note an IO port], and for whatever reason, the CHIRRTL to High FIRRTL transform only has mem.bar.en <= UInt<1>("h0").
>
> Yes. This is exactly the behaviour I see. Also, the signal is an
internally generated signal, just like in your example.

>     mem.bar.en <= UInt<1>("h1")
>
So is it just luck that the '1' gets added in this second case?

> @terpstra <https://github.com/terpstra> it'd be great to know if this was
> an issue for you before or if it's something that recently popped up.
>
The problem appeared when I rebased my tilelink2 dev branch against
rocket-chip master. That rebase spanned about three weeks worth of commits.

> I can probably try to find a way around this, but I don't really like the
> current way memory ports are inferred from the CHIRRTL. Heck, I kind of
> wish people could think a little more about revamping the Chisel Emitter,
> because, at least for memories, it feels like you have more info when
> you're coding your Chisel project, and then it's getting dropped by the
> time Chisel emits CHORTLE
>
I totally agree. As a designer, I would like to specify which ports are RW
and which clock domain a port is in. I have this information, and I'd
prefer to express it directly.

However, I think this is unrelated to the current problem. The read enable
should default to '1' not '0'...

donggyukim, 2016-09-08:
@shunshou That part is necessary to correctly infer enable signals. This is a corner case I've overlooked.

terpstra, 2016-09-08:
On Wed, Sep 7, 2016 at 11:09 PM, Angie Wang <notifications@github.com>
wrote:

> Ok, from the looks of it, even though read en was 0 in one case or 1 in
> the other case, the Verilog didn't actually use it for anything when you
> just inferred registers... but if you try with actual SRAMs, the en's
> actually need to be right...
>
Actually, it fails also in the output Verilog with inferred registers.
The read enable line there is used to conditionally update the read address
register.
The result is that the memory has the read port 'locked' at some random
initial address.

shunshou, 2016-09-08:
Oh hm maybe I just derped at reading the Verilog. @donggyukim thanks for taking a look :)

PR #289 (2016-09-14): Fix read enables for smems
* CHANGE BASE TO MASTER BEFORE MERGE
* Fixes #286

@terpstra Can you confirm this works well with your examples?
closed 2016-09-14 by donggyukim