chipsalliance/firrtl #293: FixedPoint does not currently allow binary point of zero

chick, 2016-09-10:
Following code:
```
input io_in : Fixed<6><<0>>
```
causes an error 
```
[info]   firrtl.passes.CheckChirrtl$NegWidthException$1: : [module Unit] Width cannot be negative or zero.
[info]   at firrtl.passes.CheckChirrtl$.firrtl$passes$CheckChirrtl$$checkChirrtlW$1(CheckChirrtl.scala:71)
[info]   at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlT$1$3.apply(CheckChirrtl.scala:81)
```
To see this happen, run the following in firrtl branch add-fixed-point-type
> test-only firrtlTests.fixed.RemoveFixedTypeSpec
```

sdtwigg, 2016-09-10:
It seems like this would fail for negative binary points as well?
(I am following the interpretation that this represents a UInt<width> * 2^binarypoint and thus negative binary points are reasonable.)

sdtwigg, 2016-09-10:
Actually, I just realized binarypoint is probably supposed to indicate 'where' in the UInt the binary point is so the relation is more like ```UInt<width> * 2^-binarypoint``` so decreasing the binary point while keeping the same width means you are able to represent 'larger' numbers at the expense of precision.

azidar, 2016-09-10:
Yeah, we aren't supporting negative binary points at the moment, as that would require more significant effort in changing the width inferencer. This is a minor bug, just need to allow for zero-valued binary points in the Chirrtl type checker (I already fixed the normal type checker, just missed the Chirrtl one).

closed 2016-09-13 by azidar