chipsalliance/firrtl #297: Dynamic left shift crashes firrtl in ConstProp

ucbjrl, 2016-09-13:
The following Chisel code:

    class OneHotTester(_a: Int) extends BasicTester {
      val addr = Wire(UInt(width = 32))
      addr := UInt(_a, 32)
      val t = UIntToOH(addr))
      val o = 1 << _a
      assert(UInt(o) === t)
      stop()
    }

crashes firrtl with following:

    Exception in thread "main" java.lang.IllegalArgumentException: requirement failed
    	at scala.Predef$.require(Predef.scala:207)
    	at firrtl.passes.ConstProp$.constPropPrim(ConstProp.scala:236)
    	at firrtl.passes.ConstProp$.firrtl$passes$ConstProp$$constPropExpression$1(ConstProp.scala:274)
    	at firrtl.passes.ConstProp$$anonfun$1.apply(ConstProp.scala:272)
    	at firrtl.passes.ConstProp$$anonfun$1.apply(ConstProp.scala:272)
    	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
    	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
    	at scala.collection.immutable.List.foreach(List.scala:381)
    	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
    	at scala.collection.immutable.List.map(List.scala:285)
    	at firrtl.Mappers$ExpMagnet$$anon$5.map(Mappers.scala:111)
    	at firrtl.Mappers$ExpMap.map(Mappers.scala:144)

    ...

The relevant code is:

    case Bits => e.args(0) match {
      case lit: Literal => {
        val hi = e.consts(0).toInt
        val lo = e.consts(1).toInt
        require(hi >= lo)
        UIntLiteral((lit.value >> lo) & ((BigInt(1) << (hi - lo + 1)) - 1), width_BANG(e.tpe))
      }

`hi` == -1; `lo` == 0

aswaterman, 2016-09-13:
UIntToOH is making a 2^32-bit result.  FIRRTL probably shouldn't be able to handle that, but should error in a more friendly way.

jackkoenig, 2016-09-13:
Fair point, I "fixed" the problem but then it does allow for some comically large widths. Is there an upper bound on the width of wires in Verilog?

aswaterman, 2016-09-13:
This is from a dubious source, but "The maximum port size may be limited, but will be at least 256 bits. Most software tools have a limit of 1 million bits."

In any case, it's implementation-defined.  If it's convenient, FIRRTL should feel free to set an upper bound (e.g., so that widths are always positive Ints).

jackkoenig, 2016-09-13:
This could easily be done in the apply method for IntWidth so if too large of a width is ever constructed we can error. More precise error messages can then be added elsewhere as needed.

jackkoenig, 2016-09-13:
Once #290 is merged that is

chick, 2016-09-14:
The firrtl interpreter currently has a maximum width of 100 bits, which I have not heard of it hitting.  But it sounds like it's too small by quite a bit.

aswaterman, 2016-09-14:
rocket's FPU has >1Kbit-wide signals